MarlinFirmware/Marlin #3404: Unable to do SDCard printing

sarf2k4, 2016-04-06:
I tried yesterday's rcbugfix and the rc4, I'm unable to do sdcard printing. I could see the list of gcodes but when I click that particular file the screen just goes blank and sometimes shows one weird characters. I had to roll back to rc3

Here is my config.h for rc4
[Configuration rc4.txt](https://github.com/MarlinFirmware/Marlin/files/205807/Configuration.rc4.txt)

while this one is rcbugfix
[config rcbugfix.txt](https://github.com/MarlinFirmware/Marlin/files/205806/config.rcbugfix.txt)

Sorry I didn't use the "code" markup, it has its flaws already not wrapping the whole text as "code"

turboproc, 2016-04-06:
Looks similar to a problem I had with RC5. This was resolved by using the latest Arduino IDE (1.6.5). Which IDE version did you use?

sarf2k4, 2016-04-06:
I'm using 1.0.5 because I still have sanguino 1.3, it's quite tricky or
impossible to add custom sketch to 1.6.5

thinkyhead, 2016-04-07:
For future reference, the proper way to do "code" markup is to start with 3 backticks on a line by themselves. You can follow those with the language extension. Also end with 3 backticks on a line by themselves. For example:

    ```cpp
    // Comment about the thing
    for (int i = 0; i < 4; i++) {
      // do something
    }
    ```

thinkyhead, 2016-04-07:
So far I'm unable to reproduce this issue. Perhaps I can add some special logging to the code and catch the cause of this issue for you. First I will do a comparison between RC3 and the current code to see what kind of changes have occurred around the "Print from SD" command.

Are you able to get an SD print to start if you use `M23 MYFILE.G` from a host console over USB?

sarf2k4, 2016-04-08:
I never tried the m23 command yet, I also tried with the latest arduino 1.6.8, it kinda works on sdcard printing, so basically with 1.0.6 (actual version of mine), after clicking it the screen just goes blank and I'm not sure if the printer really works. I forgot to mention that I also using standalone 1.0.6 (zip files)

I reverted back to rc3 due to potential problem with auto bed leveling (not tested thoroughly yet)

thinkyhead, 2016-04-09:
Arduino 1.0.6 seems to be the worst release, at least when it comes to compiling Marlin and all its "meta-code." So at this point we recommend almost _any release_ other than that one.

So when you say "kinda works" … how kinda does it work?

sarf2k4, 2016-04-09:
Well, i didn't test uploading the rc4/5 using 1.0.6 yet, only once. I am
running on rc4/5 now uploaded via 1.6.8 so far not much of an issue, just
that when after going to menu adjustment i somewhat stucked and gotta wait
until info screen timeout.

Is there a way to upload eagle or board of sanguino into 1.6.8 so that i am
able to upload using 1.6.8?

thinkyhead, 2016-04-09:
You should be testing `RCBugFix` because that's the latest code and bugs related to SD printing have been patched. If your problem is fixed in `RCBugFix` then we can close this issue.

I don't have any experience with Sanguino. I suggest that you ask for help at http://forums.reprap.org (for example, [this topic](http://forums.reprap.org/read.php?158,486938,639740)) and read the article http://dustsreprap.blogspot.co.nz/2015/06/better-way-to-install-sanguino-in.html

sarf2k4, 2016-04-10:
Thank you for pointing me to the blog post to add sanguino to latest arduino ide much easier, I've installed sanguino board on the 1.6.8 and it's listed on the board list, but for the time being, I'm not going to upload the rc4/5 to the sanguino board yet

closed 2016-04-10 by sarf2k4

thinkyhead, 2016-04-11:
@sarf2k4 I'm glad to hear it was helpful! We'll have to include in the Marlin documentation some of those helpful points on installing the Sanguino board for others who might also be stuck.