MarlinFirmware/Marlin #3407: termal runaway shutdown burns a hole in the build plate

mrvn, 2016-04-06:
When the firmware detects a thermal runaway and shuts down the hot end stays where it is and all the residual heat leaks into the surrounding object, or worse, the build plate leaving a mark. The hot end should be lifted maybe 5mm in Z direction before shutting down.

jbrazio, 2016-04-06:
This is a very specific condition when the thermal runaway happens near layer zero, could be an improvement if we add a bit more of logic to it i.e. lift z-pos 5mm if current z-pos lower than 10mm.

At 10mm or higher no damage will happen to the bed.

samwyze, 2016-04-06:
Personally, I'd like to see a small z lift on any thermal event. I've had a loose thermistor that caused a shutdown and melted a hole in the top few layers of the object being printed. If z had lifted it would have been possible to resume/recover the print.

jbrazio, 2016-04-07:
As long as we do not go by the physical z-max limit of the machine when lifting it should not be an issue.

thinkyhead, 2016-04-08:
I would recommend first a quick sideways move to the edge of the bed, and then a lift. Sometimes a lift straight up will leave a blob or thick string, which is almost as bad.

bdowling, 2016-05-16:
Just saying, I don't think you  can't make any assumptions about moving in XY.  Small distance then up to 80% of zmax or 10mm (whichever is greater) would be my thought.  You never know if they are working on multiple items on build plate sequentially for example, XY may not be clear sailing.

token47, 2016-07-24:
My other issue (#4191) was marked as a duplicate of this one and closed. I'm not sure it's a duplicate because, there, I talk about putting an option on the menu to quickly stop & raise (on purpose) and not just raise before a firmware freeze.

I'm leaving this comment so whoever will work on this bug please consider that other use case, please.

mkontula, 2016-08-21:
I would second the @token47's idea. There needs to be an option to have an easily and fast accessible menu item for the following sequence: 

1. Stop all movement.
2. Disable all heaters. 
3. Raise hotend at least 5mm in order to protect whatever is on top of the bed. 

When you need this functionality, your Z-leveling is off by whatever reason. You notice that filament isn't adhering and want to stop everything at once. Therefore, the Z just simply cannot remain where it is because then you would have a hole in your buildtak or even worse, smoking tape. 

Z is off by numerous reasons and at DIY level of printers, a safe "SHUT-THE-F*CK-DOWN-AND-MOVE-TO-SAFETY" is needed indeed. IMHO this should be a separate big red button at the side of your device.

Roxy-3D, 2016-08-21:
> Z is off by numerous reasons and at DIY level of printers, a safe "SHUT-THE-PRINTER-DOWN-AND-MOVE-TO-SAFETY" is needed indeed. IMHO this should be a separate big red button at the side of your device.

This really is not that difficult to do.   I added a switch to the devel_ubl branch to force the LCD Panel to go to Z-Baby-Stepping mode with zero delay.    If you are willing to wire up an extra switch (or redefine one of you buttons on the LCD Panel) I think this would be both useful and easy to do.

If nobody else gets it done within the next month, I'll do it in the devel_ubl branch for you.   (But right now there are couple of things ahead of that.)

mrvn, 2016-08-21:
@Roxy-3D While I totaly agree with you and this can use the same function I think this should be a second issue (blocked by this one).

Adding a GUI, hardware button and maybe gcode option to trigger the "abort and move to save place" function is nice to have but I don't think it is that important. If I have to trigger it manually I can also trigger heating off and move Z up 5. But when the thermal runaway triggers I'm usualy not standing next to the printer. I might not notice for 15 minutes. There it realy needs to get fixed.

Roxy-3D, 2016-08-21:
OK...   Good point...

token47, 2016-08-22:
That's ok in theory, but in practice you have to navigate through the menus to get your Z up and it takes more time than it takes to burn the bed cover. It takes at least 8 navigate+click to lift Z. The whole point here is make it quick, not to make it possible.

Roxy-3D, 2016-08-22:
Well, I think there are two points here.   There is the desire to have any thermal shut down move the nozzle to a safe location.   And the second point is to have an instant way for the user to force the nozzle away from the print instead of going through 37 layers of menus.

Incidentally, it is this second point that motivated me to add the 
```cpp
  #define QUICK_ACCESS_TO_Z_BABY_STEPPING 65	// if this option is enabled and has a pin number defined on it,
  						// it will give you immediate access to the Z-Baby-Stepping menu
						// on the LCD Panel.   This is very nice when starting a print.  You
						// need to have an LCD Panel enabled.  And you need to have a pin with
						// a switch on it to trigger the behaviour.    If you do not care about
						// having a Kill Switch on your LCD Panel, you can comment out all lines
						// in the appropriate pins_???.h file (for your machine) that have 
						// #defined KILL_PIN.   You can place that pin number here to get instant
						// access to the Z-Baby-Stepping menu.
						// Roxy's Left Switch is on pin 66.  Right Switch is on pin 65
```
option to the devel_ubl branch.   I wanted to be able to instantly use the Z-Baby-Stepping when I started a print if the height was wrong.   You only have so much time before the skirt is done being printed, and I didn't want to waste 30 seconds trying to navigate through the menu system to get there.

token47, 2016-08-22:
Yes! That kind of urgency, hehe :-)

thinkyhead, 2016-08-22:
I'm all for this kind of idea. But be aware, Marlin doesn't know that there's a "print job" running if the host is simply sending commands, so although the menu changes when you're printing from SD, we don't presume to do so when host-printing. We could assume that you're "in a print job" whenever the heaters are on, but that's about the only condition that gives a clue, and there are lots of other reasons you might have enabled the heater, such as for loading filament, or purging and cleaning the nozzle.

Also note that your LCD controller (if you have one) is equipped with a Kill Button. If you don't have it exposed, now is the time to do so! When you press the kill button it shuts down the heaters and halts the machine. It doesn't move the nozzle anywhere, however, but that could be added as a new feature.

Roxy-3D, 2016-08-22:
> Also note that your LCD controller (if you have one) is equipped with a Kill Button. 

I've always wanted more certainty.   The problem with the Kill button is the firmware has to be running and sane.   I've got a Red Switch wired up to the Reset pin of my Mega2560.

> When you press the kill button it shuts down the heaters and halts the machine. It doesn't move the nozzle anywhere, however, but that could be added as a new feature.

This thinking might not be welcome.   But if I was going to jump in and code this, I think I would have the Kill button lift the nozzle 5mm, and turn off the heaters and put a message on the LCD Panel.

But I don't think I would 'Kill' the print.  I would put it into suspended animation.   And if the person did something like 3 quick presses of the 'Kill' button, I would turn the heaters back on, prime the nozzle, and continue.   Incidentally, the user 'Prime' of the nozzle is being done on the G26 command.   I would envision something similar so the person could use this to switch filament or just make sure there aren't any holes in the print.

Tannoo, 2017-03-09:
Do it!

boelle, 2017-04-26:
this is an very old one but the idea behind it is good

maybe add it to the 1.1.1 milestone?

boelle, 2019-02-18:
@mrvn is this one still relavant?

closed 2021-12-04 by thinkyhead