MarlinFirmware/Marlin #3409: A blinking dot on the RepRapDiscount Full Graphic Smart Controller

esenapaj, 2016-04-06:
[image]
[video clip](https://www.youtube.com/watch?v=vE_WC66etvo)

It blinks at 2 second intervals on lower right hand corner of the screen.
It was tested with newest RCBugFix.
I cannot judge that it's specifications or bug or defect of my LCD, but I hasten to report it.
It looks like it haven't actual harm.

pflugj, 2016-04-06:
I only noticed now ...
Got the same blinking dot here :-)
... so your display is OK
Maybe some kind of a "running" indicator

Blue-Marlin, 2016-04-06:
Congratulations. You discovered the "I'm alive dot". It's a feature of the full graphics displays and also work in the deeper menus. As long this is blinking the refresh works.

closed 2016-04-07 by thinkyhead

esenapaj, 2016-04-07:
For @Blue-Marlin, I'm grateful to your explanation.