MarlinFirmware/Marlin #3410: MIN_POS & negative values after homing

pflugj, 2016-04-06:
Hi
...first of all: "THANKS for this nice piece of software and all your hard work!!

I am using RC5 at the moment and have some problems with the following. 
To take care of the bed offset, I did configure "**negative Y_MIN_POS**" like so
// Travel limits after homing (units are in mm)
#define X_MIN_POS 0
#define Y_MIN_POS **-30**
#define Z_MIN_POS 0
#define X_MAX_POS 200
#define Y_MAX_POS 200
#define Z_MAX_POS 300

After homing x and y manually with Pronterface the LCD shows
x:000.0 y:030.0 z:   5
y: should show "y:**-**030.0" here am I right?

The 2nd thing I noted:
My Z-Probe sensor is 30mm behind the nozzle, that is why I adjusted the Y-Endstop and therefore the Y-MIN_POS, to make sure the Probe can reach any part of the bed as well as the nozzle.
   
When issuing a G28 command I get
x:100.0 **y:055.0** z:   0.85
y should be at position "70.0" = "Probe" in centre of bed
I have enabled
 #define Z_SAFE_HOMING
    #define Z_SAFE_HOMING_X_POINT ((X_MIN_POS + X_MAX_POS) / 2) 
    #define Z_SAFE_HOMING_Y_POINT ((Y_MIN_POS + Y_MAX_POS) / 2) 

It looks like the Z-Safe home does not take care of the negative numbers 
(-30 + 200) = ONLY "170" and should be 200 in my opinion with the negative  Y_MIN_POS 
so somehow (-30+200 ) / 2 + (-30)  = **55**  is calculated here

So I think adding a "#if" condition would be the right way to correct for this:
in case of a negative "MIN_POS"  the value should be calculated as MAX_POS / 2
So I have changed the code like this:
#if ENABLED(Z_SAFE_HOMING)
   #if ( X_MIN_POS < 0 )
       #define Z_SAFE_HOMING_X_POINT  (X_MAX_POS) / 2)
       #else
       #define Z_SAFE_HOMING_X_POINT ((X_MIN_POS + X_MAX_POS) / 2)
   #endif

   #if ( Y_MIN_POS < 0 )
       #define Z_SAFE_HOMING_Y_POINT  (Y_MAX_POS / 2)
       #else
       #define Z_SAFE_HOMING_Y_POINT ((Y_MIN_POS + Y_MAX_POS) / 2 )
   #endif
#endif
This changed the G28 Home position to "Y: 070"
meaning the "Probe" is now exactly in the centre of the bed at Y=100
I think that is the correct value.

Please check if this ok.

Hey and thanks again for all your hard work
cheers
Jurgen

thinkyhead, 2016-04-07:
You can set a `[XYZ]_MIN_POS` and `[XYZ]_MAX_POS` values for movement that are greater than the size of the bed. But you will have to manually set your home positions. If you want Y to be 0 when you home, set `MANUAL_HOME_POSITIONS` and set your `MANUAL_Y_HOME_POS` to 0.

thinkyhead, 2016-04-07:
I need to contemplate this a bit more. I'll get back to you about the best settings for your setup.

pflugj, 2016-04-07:
I just read it again and according to
http://reprap.org/wiki/Calibration (look under "Build Area")
and 
http://reprap.org/wiki/Configuring_Marlin_Bed_Dimensions (using Methode 1)

To set the [XYZ]_[MIN_POS Values to a negative value looks like the correct way.
This way the machine knows where its printing area starts and I do not have to define any offsets in Slicer or Repetier host or elsewhere.
The bed dimensions are the only value needed.
By using the above mentioned changes:
The "Probe", when using  Z-SAFE HOMEING, (in my opinion) ends up exactly where it should be
"center" of the print bed.

Maybe I am wrong here but why else would we need a setting for 
[XYZ]_MIN_POS if it is "0" all the time?
or is there any another reason for this setting?

The MANUAL_[XYZ]_POS could be used to offset the REAL HOME position 

Thanks again for all helping and all your hard work

thinkyhead, 2016-04-07:
As it currently stands, you can't set a separate size for the bed, even though it may be smaller than the movement limits of the machine. If we had a separate `BED_X_SIZE` and `BED_Y_SIZE` position then `Z_SAFE_HOMING` could find the true center. Anyway, in a situation like yours, the solution is simply to set the `Z_SAFE_HOMING` to whatever works best for you and don't let it try to calculate the middle.

The `[XYZ]_MIN_POS` and `[XYZ]_MAX_POS` values specify the physical limits of each axis based on the **tip of the nozzle**. So these specify the farthest point where the **_nozzle_** can move before it physically can't. Usually, the min endstops are in the way of both X and Y, and since they determine where the zero-points are usually this makes the min XY limits both zero.

If you had the probe in front of your nozzle, you would set a higher `Y_MAX_POS` to allow the nozzle to move past the back edge of the bed.

In your case, with a probe behind the nozzle, you should set a negative `Y_MIN_POS` to allow the nozzle to move forward past "Y=0". Then you should either switch to a Y max endstop or move the min endstop so the probe can reach the front edge of the bed. If you move the endstop, then you should set your MANUAL_Y_HOME position to -30.

Unfortunately, with the current settings as they are, Marlin can't automatically figure out the center of the actual bed, or even the real size of the bed. So it does require a little more tweaking, as you have found.

closed 2016-05-04 by jbrazio