MarlinFirmware/Marlin #3424: Implementing JKN: Best way to set K and K2?

Sebastianv650, 2016-04-07:
Hello together,
I'm in the process of implementing the JKN advance feature as it is used in Sailfish. There are some things todo before I can test it on my printer, but I want to start a small discussion about how to set the K and K2 factors. Of course there should be an menu entry, but whats about a gCode? I can't find anything existing that is simmilar, also the existing advance feature in Marlin has only a fixed value in Configuration_adv.

Any proposals?

jbrazio, 2016-04-07:
Please do not let that block you, we can reserve (and implement) a specific G-Code to set the parameters just like we have to PID.

t-paul, 2016-04-07:
As Marlin user, I just wonder if that will be considered "derived from Sailfish" and therefore drag in their advertising clause from the Sailfish license.

jbrazio, 2016-04-07:
@t-paul is right.. JKN will fall into the advert clause.

Sebastianv650, 2016-04-08:
Good note about the license, I wasn't aware of that! But I think that's no problem. The official code has to show a second boot screen when JKN is enabled, but if someone is realy disturbed by it we can't block him to remove the code lines when he build his own Marlin version..

I will first fix the existing ADVANCE feature of Marlin (found the mistake(s) why it doesn't do anything visible to the print) because I think that can be done in a short time and the JKN will have the same problems in Marlin, so I can reuse something then..

jbrazio, 2016-04-08:
I doubt something requiring adverts will me merged into Marlin.. you should first gather more feedback from other devs but I will vote against adverts. :-/

Marlin is the base of a 3D printing industry and does not require anyone to advertise they were "inspired" by Marlin.

You may improve existing Marlin feature but I advise not to port Sailfish's one(s).

Sebastianv650, 2016-04-08:
I also don't like the license regarding the LCD..
I'm also asking myself what has to be done to work around that. Basicaly JKN says advance = factor * delta velocity. That's no breaking new math on it's own, so what makes a not-JKN-named feature a derivate from JKN?

I think it would be very sad not to have a (my opinion) absolutely necessary feature for high speed printing just because it's not possible to use x*dv..

Roxy-3DPrintBoard, 2016-04-08:
Can somebody clearly explain what functionality Sailfish has that we want?   At a higher level, what is this JKN algorithm and why is it good?

I wouldn't worry too much about the 'Licensing Agreement' for this and that.   We would probably just rewrite the code to run inside Marlin anyway.

thinkyhead, 2016-04-08:
Here's the relevant clause from Sailfish…

**4.0 Credits**

This software incorporates code related to acceleration from Marlin:

	https://github.com/ErikZalm/Marlin

This software is covered by GNU General Public License v3 and according to
Section(7), Subsection (b), additional permissions for author  attribution
are required on any work that incorporates, is derived or inspired from the
following components:

  1. JKN Advance
  2. YAJ (Yet Another Jerk)
  3. Altshell ReplicatorG plugin
  4. Pause @ ZPos
  5. Advance Pressure Relax

Author attribution is required as follows:

  1. If the device this software is executed on has an LCD or
     display screen attached, credit/attribution must be provided
     on this screen with at least 1 second duration when and each
     time the device is powered on.
 
  2. If the device does not have a display screen attached and the
     software is distributed in binary form, then credit/attribution
     is required to be displayed to the user prior to installation of
     the binary software.
 
  3. If the software is being supplied in source code form, then any
     existing credit/attributions must be retained.

  4. If the software is being supplied in source code form, but it is
     derived or inspired from this source code, then Credit/Attribution
     below must be provided in the source code near the top of the source
     file.

  5. The additional permissions listed here are required to be included
     in their entirety with any license file of any derivative works that
     use the above features.

Credit/Attribution:

	This software uses the following components from Jetty Firmware:
	(LIST OF COMPONENTS GOES HERE)
	authored by Dan Newman and Jetty.

thinkyhead, 2016-04-08:
Here's the thing. Marlin cannot incorporate any license that removes the freedoms given by the GPL. So, we need to look closer to see if we can even incorporate any of these elements without ourselves violating the GPL.

Sebastianv650, 2016-04-09:
@Roxy-3DPrintBoard If you print at speeds higher than jerk speed, you can see a "blob" at ever start and end point of a line in the print. It's best to see in the top solid infill of a test cube. This happens because the nozzle pressure is still high when the print head reaches it's min. junction speed to the next line. On the other hand, when accelerating to cruse speed, there is a lack of material until the nozzle pressure has settled.
A possible approach is to let the extruder run faster during accel. and slower than calculated when braking. This is the basic idea behind our (broken) advance feature and the JKN-advance. The difference between JKN and Marlin advance is the way this advance factor is calculated.
Marlin advance says advance = k * v²
JKN says = k* (vend-vstart)

Roxy-3DPrintBoard, 2016-04-09:
> Here's the thing. Marlin cannot incorporate any license that removes any freedoms given by the GPL. So, we need to look closer to see if we can even incorporate any of these elements without ourselves violating the GPL.

So any work that is 'inspired' by JKN requires attribution?   Is the term 'inspired' defined within the GPL?   Because this seems awful bogus.

jbrazio, 2016-04-09:
"Inspired" you can claim everything is inspired by something else..

Sebastianv650, 2016-04-09:
I would say i'm inspired by the blobs at the line intersection points, not by JKN ^^
A small update regarding the existing advance. The two bugs I fixed was:
.) advance_rate is calculated in planner.cpp, but never loaded from the block in stepper.cpp
.) The esteps += calculation in the ISR tries to add fractions of an integer to an integer value

Now the code is actually doing something, but not as needed that's why there is no pull request. It seems as there are even more bugs to find..
I don't want to invest time into debugging a code where I don't no if the basic math is even working, that's why I will do a linear aproach (trying not to call it JKN..) If somebody has an idea how to solve the licencse problem one day, I will provide the code.

brainscan, 2016-04-09:
When there were previous discussions about how to implement this in Marlin no one said anything at all about problems with the license. Has something changed?

CONSULitAS, 2016-04-09:
@thinkyhead 
@Sebastianv650 
@brainscan 
@jbrazio 
@Roxy-3DPrintBoard 

In general:
* Licenses are for software. This is because nearly every program / source code is automatically copyrighted. So you need a license to use it. Even if you use parts of it, you need a license.
* Algorithms, formulas, ideas or "Inspiration" is (normally) nothing you can have a copyright on. Exception: In some countries an algorithm as part of a software could be under the protection of a patent. But: IMO a "advance = factor * delta velocity" is nothing worth a patent.

After doing a little search i found:
https://github.com/jetty840/Sailfish-G3Firmware

This citation seems to be the root of the discussion, because it is the repo of "Jetty" @ jetty840 who claims for the license.

First conclusion:
* You don't need a license for the formula (or idea or inspiration), but you would need one for the source code (the implementation).
* So it is safe to write your own, but just copying could be a problem.

Lets have a closer look on the license problem, if we would just copy the code:
* They state `This software incorporates code related to acceleration from Marlin:  https://github.com/ErikZalm/Marlin`. So they copied marlin. 😄 

Now it is becoming really interesting:
* I you use marlin, you have to care about the license of Marlin: `Marlin is published under the GPL license because we believe in open development. The GPL comes with both rights and obligations. Whether you use Marlin firmware as the driver for your open or closed-source product, you must keep Marlin open, and you must provide your compatible Marlin source code to end users upon request. The most straightforward way to comply with the Marlin license is to make a fork of Marlin on Github, perform your modifications, and direct users to your modified fork.`
* And now we come to the point, why i LOVE THE GPL 😁😁😁 (for details you could google "jochen groppe gpl"). It says (one of the most funny points making it copyleft): `For example, if you distribute copies of such a program, whether gratis or for a fee, you must pass on to the recipients the same
freedoms that you received.  You must make sure that they, too, receive or can get the source code.  And you must show them these terms so they know their rights.`
* This means: If you use GPL software your new product is GPL, too. **AND YOU MAY NOT RESTRICT THE RIGHTS OF YOUR USERS**

Second conclusion(s) IMO:
* Sailfish uses Marlin and restricts the rights over GPL via "attribution". 😢 😡 So it violates the Marlin license. Funny and annoying, isn't it? 😁😁😁 
* Sailfish inherits the GPL from Marlin (by using its code) and is thereby pure GPL itself.
* So you can do anything with sailfish (via the GPLicense), if it is GPL again! 

So i would recommend to read the (whole) GPL of marlin and take special notice of the following terms:
```
  8. Termination.

  You may not propagate or modify a covered work except as expressly
provided under this License.  Any attempt otherwise to propagate or
modify it is void, and will automatically terminate your rights under
this License (including any patent licenses granted under the third
paragraph of section 11).

  However, if you cease all violation of this License, then your
license from a particular copyright holder is reinstated (a)
provisionally, unless and until the copyright holder explicitly and
finally terminates your license, and (b) permanently, if the copyright
holder fails to notify you of the violation by some reasonable means
prior to 60 days after the cessation.

  Moreover, your license from a particular copyright holder is
reinstated permanently if the copyright holder notifies you of the
violation by some reasonable means, this is the first time you have
received notice of violation of this License (for any work) from that
copyright holder, and you cure the violation prior to 30 days after
your receipt of the notice.

  Termination of your rights under this section does not terminate the
licenses of parties who have received copies or rights from you under
this License.  If your rights have been terminated and not permanently
reinstated, you do not qualify to receive new licenses for the same
material under section 10.
```

Perhaps someone of the core team should give them a "notice of violation"?

BTW: I am starting to sound like a lawyer. Shuddering... And if we really have a problem, i could perhaps ask my friend. An IT lawyer.

Just wasting time instead of translating strings. 😁

t-paul, 2016-04-09:
Sailfish probably does not violate GPL as it does say:
```
This software is covered by GNU General Public License v3 and according to
Section(7), Subsection (b), additional permissions for author attribution are
required on any work that incorporates, is derived or inspired from the
following components...
```
They use a specific section from GPL3 which seems to allow the attribution.

Which is hopefully not relevant due to the other points made about what is covered by that license.

jbrazio, 2016-04-09:
That's exactly why I suggested a "clean-room-environment" not just copying code, that would be lame.

I take from their license that they use GPL except with the advert thing for a couple of features.

thinkyhead, 2016-04-10:
Ok, I won't worry over licenses then, since the algorithm is well-known and obvious, and we will be doing our own implementation.

jbrazio, 2016-04-23:
I advise to work with @Sebastianv650 on #3589

closed 2016-04-23 by jbrazio