MarlinFirmware/Marlin #3429: After the power unexpectedly closed, open the power to continue printing

diver197, 2016-04-08:
How to achieve power down and continue to print in front of

jbrazio, 2016-04-09:
This is currently impossible, not only because it's not supported fw wise but because it will be impossible to home the robot with plastic laying on the build plate.

thinkyhead, 2016-04-09:
This is a duplicate issue. See our discussions of "WASP mode" in #2084 #2085 and #2659 

To sum up, currently it is not possible to automatically recover from a power loss, in part because Marlin doesn't do any persistent storage of the current print progress, there isn't any obvious way to detect a power failure (in order to quickly -and riskily- store the current print position before all residual power is depleted), and generally the best way to handle it requires extra hardware.

So currently if a print fails due to power outage, you must use some tricks to alter the GCode and print the remainder of the object starting just above the layer that failed. There's an article about this which you should be able to find with a Google search.

closed 2016-04-09 by thinkyhead