MarlinFirmware/Marlin #3432: Waiting to reach temperature on second tool fails sometimes

Alex9779, 2016-04-08:
Hi,
using RC4 I realized that under certain circumstances waiting to reach the temp of the second hotend fails and Marlin ends up in an endless loop.
What I do is that I have a startup script which starts to set the temperatures for bed and bot hotends using M104 and M140.
Then it auto homes and moves the head to the priming position.
When that is done I use M190 to wait for bed temp to be reached, tool 0 and 1 are still heating.
When bed is done I use M109 on tool 0 to wait for the temp to  be reached, after that M109 on tool 1.
By that time depending on what the temps are set to tool 1 might have already reached the temp.
And if this is the case then Marlin ends up in a endless loop waiting for tool 1 to reach the temp but it is already reached.
Here is some serial com protocol taken fomr an SD print:
`
Recv:  T:237.3 /245.0 B:60.0 /60.0 T0:237.3 /245.0 T1:208.0 /210.0 B@:23 @:57 @0:57 @1:55 W:10
Recv: echo:busy: processing
Recv:  T:238.3 /245.0 B:60.0 /60.0 T0:238.3 /245.0 T1:208.2 /210.0 B@:22 @:53 @0:53 @1:56 W:10
Recv:  T:239.0 /245.0 B:60.0 /60.0 T0:239.0 /245.0 T1:208.2 /210.0 B@:16 @:53 @0:53 @1:62 W:10
Recv: echo:busy: processing
Recv:  T:240.2 /245.0 B:60.0 /60.0 T0:240.2 /245.0 T1:208.6 /210.0 B@:20 @:43 @0:43 @1:58 W:10
Recv:  T:240.6 /245.0 B:60.0 /60.0 T0:240.6 /245.0 T1:208.2 /210.0 B@:14 @:46 @0:46 @1:71 W:10
Recv: echo:busy: processing
Recv:  T:241.2 /245.0 B:60.0 /60.0 T0:241.2 /245.0 T1:208.4 /210.0 B@:14 @:47 @0:47 @1:71 W:9
Recv:  T:241.6 /245.0 B:60.1 /60.0 T0:241.6 /245.0 T1:208.4 /210.0 B@:9 @:48 @0:48 @1:72 W:8
Recv: echo:busy: processing
Recv:  T:241.8 /245.0 B:60.1 /60.0 T0:241.8 /245.0 T1:208.1 /210.0 B@:11 @:55 @0:55 @1:82 W:7
Recv:  T:242.1 /245.0 B:60.0 /60.0 T0:242.1 /245.0 T1:208.4 /210.0 B@:15 @:58 @0:58 @1:78 W:6
Recv: echo:busy: processing
Recv:  T:242.3 /245.0 B:60.0 /60.0 T0:242.3 /245.0 T1:208.8 /210.0 B@:15 @:62 @0:62 @1:73 W:5
Recv:  T:242.1 /245.0 B:60.0 /60.0 T0:242.1 /245.0 T1:208.6 /210.0 B@:19 @:72 @0:72 @1:79 W:4
Recv: echo:busy: processing
Recv:  T:242.0 /245.0 B:60.0 /60.0 T0:242.0 /245.0 T1:208.4 /210.0 B@:20 @:81 @0:81 @1:85 W:3
Recv:  T:242.2 /245.0 B:60.0 /60.0 T0:242.2 /245.0 T1:208.9 /210.0 B@:23 @:80 @0:80 @1:77 W:2
Recv: echo:busy: processing
Recv:  T:242.0 /245.0 B:60.0 /60.0 T0:242.0 /245.0 T1:209.1 /210.0 B@:22 @:90 @0:90 @1:75 W:1
Recv:  T:242.0 /245.0 B:60.0 /60.0 T0:242.0 /245.0 T1:209.1 /210.0 B@:24 @:93 @0:93 @1:76 W:0
Recv: echo:Active Extruder: 1
Recv:  T:209.6 /210.0 B:60.0 /60.0 T0:241.7 /245.0 T1:209.6 /210.0 B@:19 @:69 @0:105 @1:69 W:?
Recv: echo:busy: processing
Recv:  T:209.6 /210.0 B:60.1 /60.0 T0:241.6 /245.0 T1:209.6 /210.0 B@:13 @:71 @0:110 @1:71 W:?
Recv:  T:210.3 /210.0 B:60.0 /60.0 T0:242.1 /245.0 T1:210.3 /210.0 B@:15 @:58 @0:102 @1:58 W:?
Recv: echo:busy: processing
Recv:  T:210.4 /210.0 B:60.0 /60.0 T0:242.1 /245.0 T1:210.4 /210.0 B@:17 @:58 @0:105 @1:58 W:?
Recv:  T:210.7 /210.0 B:60.0 /60.0 T0:242.3 /245.0 T1:210.7 /210.0 B@:14 @:52 @0:103 @1:52 W:?
Recv: echo:busy: processing
Recv:  T:210.9 /210.0 B:60.0 /60.0 T0:242.7 /245.0 T1:210.9 /210.0 B@:19 @:50 @0:98 @1:50 W:?
Recv:  T:210.7 /210.0 B:60.1 /60.0 T0:243.2 /245.0 T1:210.7 /210.0 B@:9 @:55 @0:91 @1:55 W:?
Recv: echo:busy: processing
Recv:  T:210.9 /210.0 B:60.0 /60.0 T0:243.1 /245.0 T1:210.9 /210.0 B@:14 @:53 @0:97 @1:53 W:?
Recv:  T:210.8 /210.0 B:60.0 /60.0 T0:243.8 /245.0 T1:210.8 /210.0 B@:16 @:55 @0:87 @1:55 W:?
Recv: echo:busy: processing
Recv:  T:211.2 /210.0 B:60.0 /60.0 T0:244.1 /245.0 T1:211.2 /210.0 B@:17 @:47 @0:83 @1:47 W:?
Recv:  T:211.3 /210.0 B:60.0 /60.0 T0:244.7 /245.0 T1:211.3 /210.0 B@:14 @:44 @0:73 @1:44 W:?
Recv: echo:busy: processing
Recv:  T:211.1 /210.0 B:60.0 /60.0 T0:245.0 /245.0 T1:211.1 /210.0 B@:20 @:49 @0:70 @1:49 W:?
Recv:  T:211.2 /210.0 B:60.0 /60.0 T0:245.5 /245.0 T1:211.2 /210.0 B@:19 @:47 @0:62 @1:47 W:?
Recv: echo:busy: processing
Recv:  T:210.9 /210.0 B:60.0 /60.0 T0:245.4 /245.0 T1:210.9 /210.0 B@:20 @:52 @0:68 @1:52 W:?
Recv:  T:210.8 /210.0 B:60.1 /60.0 T0:245.9 /245.0 T1:210.8 /210.0 B@:11 @:55 @0:61 @1:55 W:?
Recv: echo:busy: processing
Recv:  T:210.9 /210.0 B:59.9 /60.0 T0:246.4 /245.0 T1:210.9 /210.0 B@:24 @:52 @0:51 @1:52 W:?
Recv:  T:210.9 /210.0 B:59.9 /60.0 T0:246.5 /245.0 T1:210.9 /210.0 B@:25 @:50 @0:51 @1:50 W:?
Recv: echo:busy: processing
Recv:  T:210.8 /210.0 B:59.9 /60.0 T0:246.8 /245.0 T1:210.8 /210.0 B@:29 @:52 @0:46 @1:52 W:?
Recv:  T:210.4 /210.0 B:60.0 /60.0 T0:246.4 /245.0 T1:210.4 /210.0 B@:18 @:60 @0:55 @1:60 W:?
Recv: echo:busy: processing
Recv:  T:210.5 /210.0 B:60.0 /60.0 T0:246.5 /245.0 T1:210.5 /210.0 B@:16 @:57 @0:54 @1:57 W:?
Recv:  T:210.2 /210.0 B:60.0 /60.0 T0:246.6 /245.0 T1:210.2 /210.0 B@:18 @:62 @0:51 @1:62 W:?
Recv: echo:busy: processing
Recv:  T:210.3 /210.0 B:60.0 /60.0 T0:247.0 /245.0 T1:210.3 /210.0 B@:18 @:58 @0:44 @1:58 W:?
Recv:  T:210.2 /210.0 B:60.0 /60.0 T0:246.6 /245.0 T1:210.2 /210.0 B@:21 @:58 @0:50 @1:58 W:?
Recv: echo:busy: processing
Recv:  T:210.0 /210.0 B:60.0 /60.0 T0:246.6 /245.0 T1:210.0 /210.0 B@:16 @:62 @0:50 @1:62 W:?
Recv:  T:209.8 /210.0 B:60.0 /60.0 T0:245.9 /245.0 T1:209.8 /210.0 B@:17 @:67 @0:63 @1:67 W:?
[... 55 lines cut ...]
[Here I set the temp of tool 1 to 213 with the LCD and right back to 210]
Recv:  T:210.0 /213.0 B:60.0 /60.0 T0:244.7 /245.0 T1:210.0 /213.0 B@:19 @:61 @0:78 @1:61 W:10
Recv:  T:209.7 /213.0 B:60.0 /60.0 T0:245.2 /245.0 T1:209.7 /213.0 B@:18 @:106 @0:66 @1:106 W:8
Recv: echo:busy: processing
Recv:  T:209.7 /213.0 B:60.0 /60.0 T0:244.8 /245.0 T1:209.7 /213.0 B@:17 @:109 @0:75 @1:109 W:7
Recv:  T:209.8 /210.0 B:60.1 /60.0 T0:245.2 /245.0 T1:209.8 /210.0 B@:12 @:69 @0:65 @1:69 W:6
Recv: echo:busy: processing
Recv:  T:209.9 /210.0 B:60.0 /60.0 T0:244.8 /245.0 T1:209.9 /210.0 B@:19 @:67 @0:74 @1:67 W:5
Recv:  T:210.1 /210.0 B:59.9 /60.0 T0:244.4 /245.0 T1:210.1 /210.0 B@:30 @:64 @0:82 @1:64 W:4
Recv: echo:busy: processing
Recv:  T:210.2 /210.0 B:60.0 /60.0 T0:244.6 /245.0 T1:210.2 /210.0 B@:15 @:61 @0:77 @1:61 W:3
Recv:  T:210.0 /210.0 B:60.1 /60.0 T0:244.5 /245.0 T1:210.0 /210.0 B@:11 @:66 @0:78 @1:66 W:2
Recv: echo:busy: processing
Recv:  T:210.5 /210.0 B:60.1 /60.0 T0:244.7 /245.0 T1:210.5 /210.0 B@:12 @:56 @0:75 @1:56 W:1
Recv:  T:210.8 /210.0 B:60.0 /60.0 T0:244.8 /245.0 T1:210.8 /210.0 B@:19 @:50 @0:71 @1:50 W:0
Recv: echo:busy: processing
Recv: ok
Send: N3 M27*21
[...]
Recv: ok
[...]`

What I found is that the "W:" of the answer is a "?" when Marlin waits and when I set another value then it turns to a number as it is when waiting for the other hotend to heat.

I had this issue only when aborting a print and the tools and the bed were still near the target temperature when starting the new print. If the printer is cool is always works...

jbrazio, 2016-04-08:
Try RCBugfix, some patches were applied to solve bed and extruder temperature loops.

scalez, 2016-04-08:
We've discovered the bug which is causing this, if your hotend is already close to the target temperature when the while loop in M109 is entered then the _residency_start_ms_ variable will stay -1 and the loop may never escape. I've created a PR to fix it here: https://github.com/MarlinFirmware/Marlin/pull/3433

Alex9779, 2016-04-08:
Cool thanks so much!

thinkyhead, 2016-04-09:
#3433 has been merged. Please re-open if it doesn't fix your issue.

closed 2016-04-09 by thinkyhead