MarlinFirmware/Marlin #3439: Auto bed leveling does weird

JefHomeBuiltDeltaPrinter, 2016-04-09:
Hi everyone,

I have a problem with my auto bed leveling. I have a delta printer and I use the grid leveling mode. Everything seems to work if I execute the G29 code. But if a execute the G29 code and the first time he is probing I but a metal opbject underneath it, the second time also and de third time also and this works all fine. But the fourth time I don't put a metal object underneath it but he seems to detect something. From then it's everytime te same problem.

Can someone help me?

thinkyhead, 2016-04-09:
- What kind of probe?
- What version of Marlin?

JefHomeBuiltDeltaPrinter, 2016-04-10:
It's a LJ12A3-4-Z/BX Inductieve Sensor. Here you can find my marling program https://www.dropbox.com/sh/aox9c32ur6ro27d/AADR7PQ-ms98EGu08GfUXYfxa?dl=0

thinkyhead, 2016-04-11:
Please test with `RCBugFix` to see if we already fixed this.

closed 2016-05-04 by jbrazio