MarlinFirmware/Marlin #3441: Bed heating does not stop,

christianh17, 2016-04-09:
Hi!

I use a prusa 3 clone, Software: MarlinRCDevbugfix (RC5), Checkout last evening around 7 pm in Germany. Before RC5 I used RC3, there were some issues for me (bed levelling) but it worked perfect for me.

Today I had two times the problem the printer would not start printing. 

It stucks and tells me in the Log:
Recv:  T:201.8 /202.0 B:55.4 /55.0 T0:41.2 /0.0 T1:201.8 /202.0 B@:0 @:75 @0:0 @1:75 W:?
Recv:  T:201.8 /202.0 B:55.4 /55.0 T0:41.2 /0.0 T1:201.8 /202.0 B@:0 @:74 @0:0 @1:74 W:?
Recv: echo:busy: processing
Recv:  T:201.8 /202.0 B:55.2 /55.0 T0:41.2 /0.0 T1:201.8 /202.0 B@:0 @:73 @0:0 @1:73 W:?
Recv:  T:201.8 /202.0 B:55.1 /55.0 T0:41.2 /0.0 T1:201.8 /202.0 B@:0 @:73 @0:0 @1:73 W:?
Recv:  T:201.8 /202.0 B:55.1 /55.0 T0:41.2 /0.0 T1:201.8 /202.0 B@:0 @:72 @0:0 @1:72 W:?
Recv:  T:201.8 /202.0 B:54.9 /55.0 T0:41.2 /0.0 T1:201.8 /202.0 B@:0 @:72 @0:0 @1:72 W:?
Recv:  T:202.4 /202.0 B:54.9 /55.0 T0:41.2 /0.0 T1:202.4 /202.0 B@:0 @:57 @0:0 @1:57 W:?
Recv:  T:202.4 /202.0 B:54.7 /55.0 T0:41.2 /0.0 T1:202.4 /202.0 B@:127 @:59 @0:0 @1:59 W:?
Recv:  T:202.4 /202.0 B:54.6 /55.0 T0:41.2 /0.0 T1:202.4 /202.0 B@:127 @:60 @0:0 @1:60 W:?
Recv:  T:202.4 /202.0 B:54.6 /55.0 T0:41.2 /0.0 T1:202.4 /202.0 B@:127 @:60 @0:0 @1:60 W:?
Recv:  T:202.4 /202.0 B:54.6 /55.0 T0:41.2 /0.0 T1:202.4 /202.0 B@:127 @:61 @0:0 @1:61 W:?
Recv:  T:202.4 /202.0 B:54.5 /55.0 T0:41.2 /0.0 T1:202.4 /202.0 B@:127 @:61 @0:0 @1:61 W:?
Recv: echo:busy: processing
Recv:  T:201.8 /202.0 B:54.5 /55.0 T0:41.2 /0.0 T1:201.8 /202.0 B@:127 @:74 @0:0 @1:74 W:?
Recv:  T:201.8 /202.0 B:54.5 /55.0 T0:41.2 /0.0 T1:201.8 /202.0 B@:127 @:75 @0:0 @1:75 W:?
Recv:  T:201.8 /202.0 B:54.6 /55.0 T0:41.2 /0.0 T1:201.8 /202.0 B@:127 @:74 @0:0 @1:74 W:?
Recv:  T:201.8 /202.0 B:54.6 /55.0 T0:41.2 /0.0 T1:201.8 /202.0 B@:127 @:73 @0:0 @1:73 W:?
Recv:  T:201.8 /202.0 B:54.6 /55.0 T0:41.2 /0.0 T1:201.8 /202.0 B@:127 @:72 @0:0 @1:72 W:?
Recv:  T:201.8 /202.0 B:54.9 /55.0 T0:41.2 /0.0 T1:201.8 /202.0 B@:127 @:72 @0:0 @1:72 W:?
Recv:  T:201.8 /202.0 B:54.9 /55.0 T0:41.2 /0.0 T1:201.8 /202.0 B@:127 @:72 @0:0 @1:72 W:?

The print starts with 
G90
M82
M106 S0
M140 S55
M190 S55
M104 S0 T0
M104 S202 T1
M109 S202 T1

and as you cann see all temperature needs are OK (Bed 55°, Extruder 2: 202°). My workaround is: pressing reset, restarting the print-job (octoprint) and the second time it works perfect.

I remember RC4 - I tried this release to, there was the same problem for me: But there has been a bugfix. 

BTW: You all do a very very good work with marlin fw and I appreciate it very much!

Kind regards
Christian

jbrazio, 2016-04-09:
Could you please try the latest Rcbugfix ?

christianh17, 2016-04-09:
Hi!
Oh, I just noticed: I already had the last rcbugfix from yesterday evening. Are you sure there is a change inbetween?

In the meantime I did some more testing, the problem comes up when the temperature is already OK when I start printing (sending GCode). Does this help?
Kind regards
Christian

thinkyhead, 2016-04-10:
This should be patched with #3438, merged late yesterday.

jbrazio, 2016-04-10:
@thinkyhead I believe this is a different bug related with `M190` not respecting `TEMP_HYSTERESIS` as you can see from the log that the `degBed()` is 0.1C less than `degTargetBed()`. PR #3445 submitted.

christianh17, 2016-04-10:
Hi!
This morning I just tried the new rcbugfix version (last commit 562e281). Even with several attempts I could not reproduce the error.

I will try on and will give an update.

Kind regards,
Christian

thinkyhead, 2016-04-10:
Hopefully all is well now, as #3445 is not applicable here. The heater logic is supposed to keep the bed near its target temperature, and `M190` is supposed to exit its loop as soon as the bed heater turns off. We've patched that, and there is no "bed hysteresis" or "bed residency time" that Marlin needs to worry about.

closed 2016-04-10 by thinkyhead

thinkyhead, 2016-04-11:
If the stuck bed issue really was due to the temperature _never_ reaching the target, that would be a good job for the `PID_BED_DEBUG` feature. It logs the bed heating in real-time. Maybe it will expose a bug in the PID logic, or show something else about the bed itself.