MarlinFirmware/Marlin #3444: Home Offset is Applied to Mesh Bed Leveling

manianac, 2016-04-10:
After wanting to offset some Slic3r prints, I had left a home offset present with M106.  After cleaning my glass and doing a 2x2 mesh level, I found that when starting the process the firmware homes all the axis', then attempts to go to the next position honoring home offset.

In this case it causes the printer to slam into the X endstop, thankfully the belt and motor gave way.

When doing mesh leveling, shouldn't the home offset be ignored?

thinkyhead, 2016-04-10:
Yes, or well, sort of. It should add `home_offset` to the points that it's probing, so they are in the correct relative position. But this also stems from a bug in `M206`, because when you do this command it should set immediately adjust the `current_position`, but it doesn't. So, you have to do `G28` after `M206` before the current position will be correctly registered.

So, I will submit a patch for these related issues, and hopefully it will completely take care of the issue.

thinkyhead, 2016-04-10:
Thank you for pointing this out @manianac! This is a pretty important issue, even though few people know much about `home_offset`. I've also been looking at `G92` because it seems to me that we should move the software endstops whenever that command is used also. Otherwise after `G92 X1000` you can only move… left? … if at all.

manianac, 2016-04-10:
Agreed software endstops should be moved with G92 commands, otherwise you're guaranteed to drive right off the edge of your maximum.

I tried #3446 and it fixes G29, thank you!

closed 2016-04-10 by manianac

thinkyhead, 2016-04-10:
Good news @manianac … I had to do some additional changes to #3446 to account for `home_offset` in more places where the mesh bed leveling coordinates are used. It will make a difference in how the Z offset is applied across the mesh, so be sure to give it a try.