MarlinFirmware/Marlin #3447: MarlinDocumentation Live and Active

Roxy-3DPrintBoard, 2016-04-10:
@jbrazio 

I think I have the MarlinDocumentation repository live and active.   You should have 'Write' access to it.

jbrazio, 2016-04-10:
Thank you @Roxy-3DPrintBoard, I've just migrated the PoC data into the new repo and I'm waiting for the DNS replication to propagate so we can have test.marlinfw.org pointing into the official repo.

paulusjacobus, 2016-04-10:
@jbrazio well done! It looks very nice and modern. Is there a way in
collaborating on the content? I saw a few small minor glitches i.e. bed
leveling we mention to store the params into eeprom but forgot to mention
that the user has to load them in memory as well...

jbrazio, 2016-04-10:
Sure, on the repo README file I explain a bit on how to get the local dev environment going if you want to preview changes and have full control over the look and feel, but in a nutshell proceed the same way you would do it Marlin, fork the repo, spawn a new branch, edit the `articles/` and commit a new PR against `master`.

Roxy-3DPrintBoard, 2016-04-10:
@jbrazio    Are all of the permissions appropriately for you to make quick progress on MarlinDocumentation?   If you need a change, please just holler!

jbrazio, 2016-04-11:
So far everything looks good. I've just two picky things to ask as I cannot enter the settings page for the repo:
 - disable the wiki for this repo
 - add the URL http://marlinfw.org on the appropriate field near the repo description

I hope now people will "trust" this repo and start contributing, so far not being inside the MarlinFirmware org seemed like a blocking point for some.

jbrazio, 2016-04-11:
@Roxy-3DPrintBoard I can't also activate Travis, I can see the repo but travis complains:

> You require admin rights to enable these repositories

Roxy-3DPrintBoard, 2016-04-11:
OK...  Let's try it now!

jbrazio, 2016-04-11:
Thank you Roxy !

jbrazio, 2016-04-12:
We are now live at http://marlinfw.org.
I've added a button to every page "Improve this page".

closed 2016-05-06 by jbrazio