MarlinFirmware/Marlin #3448: Endstop status on LCD

sarf2k4, 2016-04-10:
I would like to suggest an endstop status on LCD for each axis for min and max.

On Reprap discount smart controller, there are extra unused block on the head position line (line 2) and at the end of the each axis block would display at least single line. When it is open means it is not present.

For example of the display of the head position are displayed like this `X  0 Y  0  Z000.00 ', the block that I'm pointing out is marked with `*` on these `X  0*Y  0 *Z000.00*'. Basically these are the place to indicate the endstop state. Bottom lines are for mins, top are for max.

Even better if 2 line of each block can be used to indicate even more specific status since in the config has `use_software_ensdstop`; single line = software endstop, double = physical endstop. 

This is good for troubleshooting without ever using the laptop just to check whether or not the endstop really works. I would suggest to make this as optional feature for user to turn on

thinkyhead, 2016-04-11:
I assume you mean the Graphical controller (where we can do such tricks). It would be a good add-on feature for troubleshooting.

sarf2k4, 2016-04-11:
I would like to add a little bit of more info on this endstop status indicator, since the min and max will be displayed at the bottom and top of the block, what I meant about single and double line are;

first pixel row of bottom and top are physical endstop, while the second row of (from) top and bottom are software. Or vice versa.

So if both of them are triggered, 2 rows will lit up. for example in pseudo code below

```
if (physical max = triggered)
    first row lit up only
if (software max = triggered)
    second row lit up only
if (software max, physical max = triggered)
    first and second row lit up at once <= This one is when the axis really do match with the defined max travel when the endstop has been triggered
if (head position out of software endstop range = true)
    second row will blink <= e.g homed at x-2, y-5
```

closed 2016-05-10 by jbrazio