MarlinFirmware/Marlin #3449: Change filament on LCD back after silk buzzer example, according to the LCD cannot continue

liangruiben, 2016-04-10:
What I use is Chinese arduino uno r3, the new batch have this problem, after I check find numerical here is too small, I modified normal, don't know if other people have encountered this problem, I just put forward my problems and solutions, I confirm that will not encounter this problem is one of the few

inline void gcode_M600() 
.......................
    while (!lcd_clicked()) {
//-----------Change the following sentences---------------------------
      if (++cnt == 10240){ lcd_quick_feedback(),cnt=0;} // every 256th frame till the lcd is clicked
      manage_heater();
      manage_inactivity(true);
      lcd_update();
    } // while(!lcd_clicked)

CONSULitAS, 2016-04-10:
Without seeing the deltas i don't understand what you changed and why.

Please explain the problem in simple sentences.

Please make a pull request with github. Thanks.

liangruiben, 2016-04-10:
我是GIHUB新手，所以不太懂操作
I am a GIHUB novice, so don't know much about the operation

我修改了++cnt == 10240，
I have modified the + + CNT = = 10240,

原来的设置太快，使得检查按键失败
The original set up fast, make check button failure

liangruiben, 2016-04-10:
如果你不能理解，我可以给你录一个视频重现这个错误
If you can't understand, I can give you recorded a video is this mistake again

CONSULitAS, 2016-04-10:
for GitHub read this (and others):
https://guides.github.com/activities/hello-world/

liangruiben, 2016-04-10:
好的，我会学习如何使用这个功能，非常感谢您
Ok, I'll learn how to use this feature, thank you very much

liangruiben, 2016-04-11:
The RC4 version of the latest code seemed to have fixed the problem

thinkyhead, 2016-04-12:
Thanks! Be sure to always test with the latest code (in this case `RCBugFix`) to see if we already fixed a bug. Otherwise we chase gooses, our tails, and castles in the air.

closed 2016-04-12 by thinkyhead