MarlinFirmware/Marlin #3450: I now have UM1 3d print, How to implement G28 to double Z min endstop

liangruiben, 2016-04-10:
I now have UM1 3d print, How to implement G28 to double Z min endstop 
I try to GCODE solved, however
G28
G1 Z10
G28
...
But the actual G1 Z0 no execution

thinkyhead, 2016-04-11:
In filing a bug report, please provide…
- Describe the Expected Behavior
- Describe the Actual Behavior
- List the Steps Required to reproduce the issue

liangruiben, 2016-04-11:
Describe the Expected Behavior
	Z axis AUTO HOME, move down 20 mm, AUTO HOME again

Describe the Actual Behavior
	The Z axis came home didn't move down

List the Steps Required to reproduce the issue
	Save GCODE to SD card, select PRINT FROM SD to PRINT
//------------------------------
M107
G28
G1 Z20
G28
G1 Z15.0 F5400;Move the platform down to 15 mm
G92 E0.Zero the extruded length
G1 F200 E3.Extrude 3 mm of feed stock
G92 E0.Zero the extruded length again
M104 S0.Extruder heater off
M140 S0.The heated bed heater off
M84;Steppers off
G90;Absolute positioning
//------------------------------

jbrazio, 2016-04-11:
> G28
G1 Z20
G28

You're homing twice.. this does not work as expected.
From you comment can I assume you want to apply a 20mm offset to the "first" home postiion ?

liangruiben, 2016-04-11:
不是，是因为如果打印时平台刚好在ZHOME时，G28时平台会有偏移
NO, because if the print platform just in Z HOME, G28 platform will have a deviation

重现方法
Repeat method

G28，把平台向下移0.2mm，当限位还没有复位时，再G28位置不正确
G28, the platform down 0.2 mm, when the limit is not reset, then G28 position is not correct

现实中出现的时机
In the reality of the time

打印第一层失败时重打就会有很大几率出现
Print the first layer failure retype it will have a great chance

jbrazio, 2016-04-11:
If you have an offset set with `M851` (do `M501` to see if you have).
When you do G28 to home, the position of Z will be equal to the offset value set.

liangruiben, 2016-04-11:
可能我翻译的意思不正确
May be the meaning of my translation is not correct

因为打印机的机械限位开关会有1mm左右的接通距离（+-1mm都在接通状态），如果G28时，刚好在这个区域，G28不会动作，实际偏移与调平时的偏移有差值
Because the printer mechanical limit switch will have 1 mm or so on distance (+ - 1 mm is in the on state), if G28, just in this area, G28 not action, the actual migration and usually there's offset value

CONSULitAS, 2016-04-11:
@liangruiben 
> May be the meaning of my translation is not correct

Perhaps find a english speaking human to help you translating.

Robots are not good in translating complex problems.

liangruiben, 2016-04-11:
[image]

liangruiben, 2016-04-11:
如上图所示
As shown in the above

平台在1MM范围内都会被机器认为归零
Platform within 1 mm will be considered by machine to zero

liangruiben, 2016-04-11:
所以我希望先把平台降下后，再归零，
After the rain, so I hope the first platform to zero,

因为必须选归零才能移动平台，所以
Because must choose zero to mobile platforms, so

G28
G28

G1 Z10
G1 Z10

G28
G28

但是不起作用，归零后没有向下移动10mm
But don't work, didn't move down after zero 10 mm

jbrazio, 2016-04-11:
Do `G28` then take a ruler to measure distance from nozzle to bed then do:

```
M851 Z-1.0
M500
G28
```

Where `-1.0` is the distance you have measured (must be negative).
Please watch out for the nozzle not to crash on the table.

liangruiben, 2016-04-11:
你这样是不起作用的，如果再次归零，他会移动原来的位置上
You this is not effective, if zero again, he will move the original position

jbrazio, 2016-04-11:
When you have `M851` set it `G28` will respect your offset.

liangruiben, 2016-04-11:
1, G28
[image]

2 G1 Z1
[image]

3 G28
[image]

平台没有移动，和第一次的G28位置不同
No mobile platform, G28 position different from the first

liangruiben, 2016-04-11:
如果移动更多，再次G28，他会回到第一次的位置上
If more mobile, G28 again, he will come back to the first position

liangruiben, 2016-04-11:
请注意，机械限位开关复位会有一个很小的距离
Please note that the mechanical limit switch reset will have a very small distance

liangruiben, 2016-04-11:
测试代码
The test code

G28
G1 Z0.1

如果你的机械限位开关有指示灯，你会发觉他没有关闭
If you have light mechanical limit switch, you'll find he did not shut down

liangruiben, 2016-04-11:
如果这个时候你打印，你会着床失败
If you print this time, you'll be implantation failure

thinkyhead, 2016-04-12:
The `M851` value (default set by `Z_PROBE_OFFSET_FROM_EXTRUDER`) should measure the distance from the NOZZLE to the BED at the precise moment (or point) when the PROBE switch is triggered. You may find with `M48` that your probe is not very reliable.

liangruiben, 2016-04-13:
探针激活后，下移0.2mm还在激活状态，如果这时ABORT PRINT，重新打印，G28的位置就不正确了
Probe activated, down 0.2mm also in the activated state, if ABORT PRINT at this moment, to PRINT, G28 position is not correct

需要向下移10mm或更多，再重新G28才正确
Need to move down 10 mm or more, G28 again is right

jbrazio, 2016-04-13:
@liangruiben I'm really sorry.. but it's very hard to understand yout *exact issue* and what you want to archive. Do you think it's possible to explain it in a video ? I really do not know if it could help.

liangruiben, 2016-04-14:
OK,I recorded a video to show yo

closed 2016-05-06 by jbrazio