MarlinFirmware/Marlin #3451: Why can't achieve in LCD setting Z_MAX_POS

liangruiben, 2016-04-10:
Because I am too much machine height is different, each time to refresh the firmware need to modify the source code to upload at a time

thinkyhead, 2016-04-11:
Can you rephrase? What do you want to do?

liangruiben, 2016-04-11:
我有很多台高度不一层的3D打印机，可以在LCD里实现修改Z_MAX_POS的参数吗？
I have a lot of highly mixed layer 3d printer, can modify Z_MAX_POS parameters in the LCD?

这样我就不需要针对每台3D打印机来修改固件的参数值。
So I don't need for every 3d printer to modify the firmware of the parameter values.

thinkyhead, 2016-04-12:
With Marlin, every printer with different physical characteristics (height, width, endstops, etc.) must be flashed with different `Configuration.h` and `Configuration_adv.h` parameters. The physical characteristics of a single printer very rarely change, so there is no need to make items like `Z_MAX_POS` into a variable. It's not too hard to maintain multiple configurations, especially if they are similar.

liangruiben, 2016-04-12:
我有超过50台3D打印机。。。
I have more than 50 units 3d printer...

每次更新固件都需要单独配置。。。
Each update firmware requires separate configuration...

这需要花上1天的时候
This need to spend 1 day

liangruiben, 2016-04-12:
好吧，我承认对于普通人来说这不是一个必要的功能
Ok, I admit that for the average person this is not a necessary function

jbrazio, 2016-04-12:
@liangruiben this is something will not be implemented into the current Marlin, but we may have this into consideration for MarlinDev, people maintaining large print facilities could have an easier life.

@thinkyhead please mark this one as Feature Request.

closed 2016-04-23 by jbrazio