MarlinFirmware/Marlin #3452: Trouble configuring Reprap LCD

outbound-pt, 2016-04-10:
Greetings,

I have a 20x4 LCD for use with the Megatronics v3. It's supposed to be plug and play for the Megatronics as it has come fully assembled. The Reprap keypad and SD card reader work perfectly fine, but not the LCD.

I've tried following the tutorial but I'm no being able to properly configure the display. All I get at most is two lines of solid squares when I mess around with the trimpot on the board. I do get a few characters on the top line right now.  If I move the trimpot at most I'll get the first and third line all solid squares.

Is it possible that my LCD is incorrectly assembled (I bought it fully assembled from the store)? 

I asked about it in the Reprap forums but so far no one has been able to assist me.
http://forums.reprap.org/read.php?415,637038,650821

Thank you for any help you might provide.

jbrazio, 2016-04-10:
Did it came with the firmware installed it you had to compile your own ?

outbound-pt, 2016-04-11:
I had to compile my own. I took the generic Marlin Firmware and configured it for use with Megatronics v3.

jbrazio, 2016-04-11:
You should use RCBugfix branch and the changes required to have the display working should be done on `Configuration.h` and/or `Configuration_adv.h`. Could you please attach by drag and dropping into this thread both files ?

outbound-pt, 2016-04-11:
I'm having trouble uploading the files but it's on [Google drive](https://drive.google.com/folderview?id=0B2Mxvzx_VOx3Mm16S192N09zbjg&usp=sharing)

Supposedly anyone with the link can access the folder with the firmware on the board.

FYI, everything else is working fine. Also, there are a couple of conflict files (due to the Google Drive software), but if I'm not mistaken the ones on the board are the non conflict ones.

jbrazio, 2016-04-11:
Antonio Soares ? Depois de ver o Google Drive é que o "-pt" fez sentido. :-)
Devias actualizar o teu Marlin para a última versão, a que tens no Google Drive é relativamente antiga, podes descarregar o zip [daqui](https://github.com/MarlinFirmware/Marlin/archive/RCBugFix.zip).

outbound-pt, 2016-04-11:
Obrigado (Thanks).

I guess I'll have to use a more recent version of the firmware and reconfigure everything else once more for use with my Leapfrog Creatr (I was trying to avoid that).

jbrazio, 2016-04-11:
But what did you had to change exactly to get it working ? Maybe we can introduce those changes into Marlin.

outbound-pt, 2016-04-11:
It was just specific configuration settings for my version of the Leapfrog Creatr Dual (version 2b), which I replaced the native controller with Megatronics.

Because the Leapfrog printers use Marlin firmware it's just a matter of copying and pasting a lot of the settings.

jbrazio, 2016-04-11:
Do not just copy paste the settings a lot of `#define` have changed, so you have to be little careful when building the RCBugfix conf file. And do not change anything on any file other than the two config files.

outbound-pt, 2016-04-11:
I just copy and paste the settings regarding the hardware setup (thing like extruder head values, endstop, etc.). That and a small change I did to the stepper motor pins in order to switch the Y and Z axis outputs since my printer has two stepper on the Y axis. Anyway, the Reprap guys told me I just likely have the cable reversed. I'll have to try that. :D

outbound-pt, 2016-04-11:
Still, nothing, all I get is a blue screen with the cables reversed...

jbrazio, 2016-04-11:
Sorry to be playing the same key but this is important, you're now compiling and running RCBugfix ? If so could you please [zip] drag and drop here the two config files ? Thanks.

outbound-pt, 2016-04-11:
Nope, I haven't had time to configure that one.

Later I'll be trying directly with the Reprap firmware configured directly for Megatronics v3. It was in their software section (I have no clue why they just didn't put it in the Megatronics page along with all the other board info and Test firmware) and I'll try that.

jbrazio, 2016-04-11:
If you want our help you gotta start trying what we ask you to try/do.

outbound-pt, 2016-04-11:
Yes, sorry I was also having some correspondence with the Reprap technical support. He suggested I try the Marlin Firmware that's already configured for Megatronics v3 because the LCD support is already enabled and the correct pins configured. Once I have time I'll try that one and see if that's the problem or if it's a hardware issue with the cables. I'll keep you posted.

jbrazio, 2016-04-11:
Great. I understood you bought the robot as a kit, if the vendor provides you a "working firmware" and your hardware is not working then I would assume faulty hardware.

outbound-pt, 2016-04-11:
Actually the Leapfrog is sold as a fully assembled and calibrated machine. Their board is proprietary but basically still a Reprap, just their own PCB design. I shorted the first board and the replacement they sent me was faulty. To replace it again I would have to purchase one but I was already so fed up with the company that I decided to just replace it with an open-source alternative, hence the Megatronics which I purchased off RepRapWorld. I didn't get the firmware off of their website as I couldn't immediately find it and wherever I got it from I guess I ended up downloading an out of date version. Hopefully the correct firmware will solve this issue.

Roxy-3DPrintBoard, 2016-04-11:
> He suggested I try the Marlin Firmware that's already configured for Megatronics v3 because the LCD support is already enabled and the correct pins configured. 

The source code will be available for what they ship.   If you take their configuration.h file and look to see how they have the LCD Panel defined...   You should be able to cross those settings over to the RCBugFix configuration.h and have it work...

thinkyhead, 2016-04-12:
The settings as configured in https://github.com/MarlinFirmware/Marlin/blob/RCBugFix/Marlin/example_configurations/RepRapWorld/Megatronics/Configuration.h should be very close to what you're looking for. Enabled options include:
```cpp
#define ULTRA_LCD  //general LCD support, also 16x2
#define SDSUPPORT // Enable SD Card Support in Hardware Console
#define SD_CHECK_AND_RETRY // Use CRC checks and retries on the SD communication
...
#define REPRAPWORLD_KEYPAD
#define REPRAPWORLD_KEYPAD_MOVE_STEP 10.0 // how much should be moved when a key is pressed, eg 10.0 means 10mm per click
```

jbrazio, 2016-04-14:
@outbound-pt what is the outcome of your problem, can we still help you in some way or can we close this issue ? Thanks.

outbound-pt, 2016-04-14:
My apologies, I wasn't aware of how the messaging here worked.

I haven't had the time to see if the solutions given would work but since I'm not sure when I'll be able to I think it's best to close it then.

Thank you all.

jbrazio, 2016-04-14:
Thanks @outbound-pt, if you need just re-open the ticket.

closed 2016-04-14 by jbrazio

outbound-pt, 2016-05-19:
HI again,

I was finally able to try out both the supplied firmware and the latest RC bug fix. I'm still getting the same problem.

I will try with a smaller 16x2 display I have from an arduino kit, but I suspect that the assembled 20x4 might have some problem with it.

thinkyhead, 2016-05-19:
Sounds like a bad piece of hardware. Perhaps it's fixable, as it might only be one bad component on the board.

outbound-pt, 2016-05-19:
Would [this one](http://www.ptrobotics.com/lcd-grafico/1027-graphic-lcd-128x64-stn-led-backlight.html) work?

Blue-Marlin, 2016-05-19:
KS0108B is supported by u8glib. But for the same prize you can have a complete panel - for example a "REPRAPDISCOUNT FULL GRAPHIC SMART CONTROLLER" clone.

outbound-pt, 2016-05-19:
I think you mean price, but for my board the one you mention will cost me more. This one is for the price listed on the website, I just pick it up from the store. The [clone for the Megatronics](https://reprapworld.com/?products_details&products_id/1218/cPath/1735_1626#.Vz3tReSLdpU) costs at least €40 plus shipping from the Reprap store in the Netherlands (I'm in Portugal).

The only thing I like about that controller is the screen anyway, I'd rather have the Reprap control panel with the directional keypad.

I think I'll opt for this screen and try and make it work with my current setup. Too bad as I had already [modified a Thingiverse project](https://cad.onshape.com/documents/765a9d89f1ca6bd3a97c1ebd) to have this particular setup mounted on the machine.