MarlinFirmware/Marlin #3458: Servo only (no bed leveling) and z endstops 1.1.0 RC4

spanner888, 2016-04-11:
I am trying to implement use of one servo only without using bed leveling. The servo is being used for an entirely different purpose, so the default z_endstop is required to function as normal.

However in Configuration.h, uncommenting below and setting number of servos to 1
    #define NUM_SERVOS 1
seems to disable z_endstop, which can lead to head crash into bed.

I have successfully configured auto bed leveling with a z probe as a test, but as noted above, the servo is being used for a different purpose, so bed leveling is not required, and in fact bed leveling will interfere with my usage.

My initial look at the Marlin code seems to suggest the above behaviour is by design, ie #define NUM_SERVOS 1, leads to code like
    #if defined(NUM_SERVOS) && NUM_SERVOS > 0
.....
    #define HAS_SERVO_ENDSTOPS true
    #define SERVO_ENDSTOP_IDS { X_ENDSTOP_SERVO_NR, Y_ENDSTOP_SERVO_NR, Z_ENDSTOP_SERVO_NR }

**My question is:- is it possible to use one or more servos AND keep using the normal z_endstop (and not use any of the auto level code/endstops)?
**
Background:-  Firmware = Marlin 1.1.0 RC4, configured & working with Sunhokey Prusa i3

Blue-Marlin, 2016-04-11:
With some luck
Conditionals.h
```cpp
@@ -704,11 +704,11 @@
       #define SERVO_ENDSTOP_IDS { X_ENDSTOP_SERVO_NR, Y_ENDSTOP_SERVO_NR, Z_ENDSTOP_SERVO_NR }
     #endif
   #endif
 
   #if ( (HAS_Z_MIN && ENABLED(Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN)) || HAS_Z_PROBE ) && \
-    ( ENABLED(FIX_MOUNTED_PROBE) || defined(Z_ENDSTOP_SERVO_NR) || ENABLED(Z_PROBE_ALLEN_KEY) || ENABLED(Z_PROBE_SLED) )
+    ( ENABLED(FIX_MOUNTED_PROBE) || (Z_ENDSTOP_SERVO_NR >= 0) || ENABLED(Z_PROBE_ALLEN_KEY) || ENABLED(Z_PROBE_SLED) )
     #define HAS_Z_MIN_PROBE
   #endif
 
 #endif //CONFIGURATION_LCD
 #endif //CONDITIONALS_H
```
is all we need.

spanner888, 2016-04-11:
No luck required - works well, thanks and also thanks for super quick reply.

I would also like to suggest this becomes a Marlin standard feature, as more & more people are now using 3D printers (with servos) for a wide variety of other uses that do not require use of servos to be tied to auto bed leveling.

jbrazio, 2016-04-11:
We will keep that on our mind for the next gen of Marlin. Thank you for the suggestion.

closed 2016-04-11 by spanner888