MarlinFirmware/Marlin #3459: MAXTEMP triggered when requesting Preheat ABS Both too soon after boot

snowzach, 2016-04-11:
When I select Preheat ABS Both very quickly after the firmware boots it will trigger a MAXTEMP error and hang. If I select Preheat ABS 1 and then 2 it seems fine. If I wait for 15-20 seconds before I select Preheat ABS Both it works okay.

Blue-Marlin, 2016-04-11:
What sensor type do you use?
Is `SDSUPPORT` compiled in and a SD-card present?
How do you send your commands? [pannel, auto connection script from host, autostart file from sd ]
What version/branch are you on.

Depending on the sensor type it needs some time to get stable temperature readings.
The processor can be really busy fore some seconds if it has to read a big directory from sd. That may delay stable temperatures further.
If you are sending the temperature settings in a script, a tiny delay (M4 S20) may help.

snowzach, 2016-04-11:
I am using a 100k thermistor. 
I am using an SD Card and it is present. It does have a good number of files on it as well.
I use the LCD Panel to select the Preheat ABS Both command.
I am currently on the latest of the RC branch. 

I was about to go try the RCBugFix branch as I'm having issues printing as well but haven't narrowed down the issue yet. (When it goes to print second color, Z axis just starts moving upwards and I shut it off before it runs off the rails) I will test this issue on that branch also and report back.

snowzach, 2016-04-11:
So it would appear that this issue is resolved in the RCBugFix branch... Now I just need to figure out why it picks a Z direction and runs forever at a certain point in my print. I will open another issue if I don't figure it out as I've not seen this in 1.0.2.

snowzach, 2016-04-11:
Yahhh... So I've been testing some more and the issue is definitely still there in RCBugFix. If I wait a bit, it won't trigger. It seems pretty sensitive.

thinkyhead, 2016-04-12:
@snowzach That sounds like a nasty bug. Please open an issue specifically for that one.

thinkyhead, 2016-04-12:
@snowzach When you write your new issue, please include what kind of printer you're using. Delta and CoreXY printers may show different issues than Cartesian robots.

thinkyhead, 2016-04-12:
I see it's there now - #3461

closed 2016-04-12 by thinkyhead