MarlinFirmware/Marlin #3460: System hangs when printing from SD

lakester, 2016-04-11:
When I print from SD, upon selecting the file, the control panel goes blank and the system hangs.  If the system was preheating beforehand, is APPEARS (judging from fan speed on the power supply) that all heating elements are shutdown upon hanging.

I had heard mention of a problem with files beginning with a number, and it so happens that was the case with the file I was trying to print, but in fact the hang occurs with any file.  I've already pulled over the latest RCBugFix which is thought to fix that problem (the number thing), but the problem persists (the hang on prints of any file).

The firmware:

Marlin 1.1.0-RCBugFix

The hardware:

* Makerfarm i3v 12"
* RUMBA
* "Standard" (non-graphical display)

Misc related defines from Configuration.h:

#define MOTHERBOARD BOARD_RUMBA
#define SDSUPPORT
#define REPRAP_DISCOUNT_SMART_CONTROLLER
#define EEPROM_SETTINGS

t-paul, 2016-04-11:
Which Arduino version did you use to compile? Maybe it's related to the issue I had: #3229.

lakester, 2016-04-11:
Heh..., I'm chuckling.  That was actually on my list of things to wonder about..., but it was dead last on my list ;).  I'm running Arduino 1.0.6.

OK..., I'll play around with what version of Arduino at or beyond 1.5 I can get working on my Mac and talking to the board.

Thx!

lakester, 2016-04-11:
Alrighty!  Installed Arduino 1.6.8 on the Mac..., rebuilt and installed RCBugFix..., and all appears to be well now.  I didn't have time to let the print actually complete..., but it did correctly begin execution of the file by starting the heating process, etc..., so I'm gonna say the rest of the print will probably work as well.

Closing this issue..., thx again!

closed 2016-04-11 by lakester