MarlinFirmware/Marlin #3461: Z starts moving indefinitely during print

snowzach, 2016-04-11:
I've got a dual extruder setup I've been working on and trying to use the latest RC of Marlin to do it. I've got a problem with this print where it will start moving on the Z-Axis and it doesn't seem to stop in the middle of the print. I've tried with RC5 and the RCBugFix build and they both do it but oddly the RC5 moves Z axis up and the RCBugFix moves Z-Axis into the bed. It seems to work okay on Marlin 1.0.2.

I've attached the GCODE and the Config file. I marked the line: ; HERE IS WHERE IT APPEARS TO DRIVE Z INDEFINITELY in the GCODE file.

This is a pretty run of the mill setup as far as I know.  The only thing that is different is that it does use an IR Z Probe so it's triggered most of the time during the first few layers. I have FIX_MOUNTED_PROBE set.

[Configuration.txt](https://github.com/MarlinFirmware/Marlin/files/213513/Configuration.txt)
[TheGCode.txt](https://github.com/MarlinFirmware/Marlin/files/213514/TheGCode.txt)

snowzach, 2016-04-11:
I tried printing the file from an SDCard and using Pronterface with the same results.

snowzach, 2016-04-11:
So this is shockingly easy to reproduce now...
; Warm em up
M104 S230 T0
M104 S230 T1
; Switch to extruder 1 and retract
T1
G1 E-2.00000 F2400.00000
My Z-Axis just starts moving upwards and I presume it's not going to stop

When doing it inside of a print, the first 3 layers print perfect until it switches extruders and tries to retract.

thinkyhead, 2016-04-12:
What kind of machine are you using? Cartesian, Delta, CoreXY, or CoreXZ?

thinkyhead, 2016-04-12:
Once #3476 is merged you can use `DEBUG_LEVELING_FEATURE` to get additional output for this issue, which will show what's going on in that code block. You will need to use `M111 S32` to enable logging of this info. From that output we should be able to figure out what's going on here.

snowzach, 2016-04-12:
I'm using a Cartesian with an IR Z bed probe and ramps hardware.

snowzach, 2016-04-12:
So I'm pretty sure I fixed the issue...  

Basically the extruder_offset array is initialized with XY values unless you have a dual X carriage machine and then it is initialized with XYZ values... However when using bed leveling it also needs XYZ values. When it applied the offset it was overrunning the array.

I changed the definition to:

```
// Extruder offsets
#if EXTRUDERS > 1
  #ifndef EXTRUDER_OFFSET_X
    #define EXTRUDER_OFFSET_X { 0 }
  #endif
  #ifndef EXTRUDER_OFFSET_Y
    #define EXTRUDER_OFFSET_Y { 0 }
  #endif
  float extruder_offset[][EXTRUDERS] = {
    EXTRUDER_OFFSET_X,
    EXTRUDER_OFFSET_Y
    #if ENABLED(DUAL_X_CARRIAGE) || ENABLED(AUTO_BED_LEVELING_FEATURE)
      , { 0 } // supports offsets in XYZ plane
    #endif
  };
#endif
```
Seems okay now...

thinkyhead, 2016-04-12:
@snowzach That's a really good catch. I was just looking at that array too, and although it looked strange to me I couldn't pin it down. I will apply the change and merge it asap!

snowzach, 2016-04-13:
Pulled and tested. Looks good!

closed 2016-04-13 by snowzach