MarlinFirmware/Marlin #3463: Is it possible to force the use of Arduino v1.6.8 ???

Roxy-3DPrintBoard, 2016-04-11:
People are continuing to use the older versions of Arduino that we know break the SD-Memory card code.   Can we add a Sanity Check or some other mechanism to alert the user they are using the wrong compiler?

http://3dprintboard.com/showthread.php?21439-About-to-try-Marlin1-1-0RC4&p=85625&viewfull=1#post85625

jbrazio, 2016-04-11:
Yes we can force 1.6.x, I submitted a PR to address this issue today.

CONSULitAS, 2016-04-11:
#3462

CONSULitAS, 2016-04-11:
@jbrazio 
Just a second slower. 😦

fbarcenas, 2016-04-12:
With version 1.6.8 i have problems with the SDCARD file

With 1.6.7 version, the problems is totally solved.
I only want say this, for not force the people to 1.6.8

jbrazio, 2016-04-12:
humm interesting.
I had a look https://github.com/arduino/Arduino/compare/1.6.7...1.6.8 and there are some changes with ISR and SDCard lib, but AFAIK Marlin does not uses the "internal" SDlib of Arduino.

Roxy-3DPrintBoard, 2016-04-12:
@fbarcenas 

We never did get to 'Root Cause' of why certain versions of Arduino create sick versions of Marlin.  It could be the libraries included with Arduino.  But it also could be a problem in the GCC compiler.  

This would be a valuable thing to know and fully understand.  But it would take a lot of effort to get the answer and right now nobody has the time (and capability) to do it.   So until we get this fully resolved, we probably should just stick to the 'Known Working' versions of the compiler.  

But...   I have a quick question...  Is this documented anywhere besides having the source code blow up when you try to compile it?   I can't find any other reference to this requirement.

paulusjacobus, 2016-04-12:
Hi Roxy my compilers 1.6.8 and 9 do compile properly so it must be somewhere in the libraries. I did a clean install, so removed the old ide first.

Roxy-3DPrintBoard, 2016-04-12:
But did you get the failure when you compiled using the other versions?

fbarcenas, 2016-04-12:
I never remove everything(with folders library and hardware) when i install a new version of arduino. Can be this a problem?
A stupid question....: i use mac version...can do more or less problem for use win or mac version??

paulusjacobus, 2016-04-12:
@Roxy no I never got the error before. I actually had to move to 1.6.8 on
one of my laptops in order to flash the printrbot Rev F board for a friend
(requires teensy2+ drivers and programmer, so a daunting task). So I really
wonder if old IDE files when left on the PC causes the issue. I never tried
to install multiple versions on one laptop to see if it still compiles. I
will try this out during weekend after testing RCBugFix on my CoreXY
(request from Scott)

CONSULitAS, 2016-04-12:
@fbarcenas 
I recently had the problem, when updating Arduino (on my Mac), that the Liquid Chrystal library did not work. Compiler errors.

After searching and cursing the IDE i found out that in the newer version the library is delivered as a standard library in the package. For the older IDE i still had the library in my user/Arduino/libraries/ folder. But this was an old version and not compatible.

Removed the old library and everything was fine again.

So: Do some housekeeping after upgrades and be sure that there are no old libraries which could disturb.

Roxy-3DPrintBoard, 2016-04-13:
> Removed the old library and everything was fine again.

> So: Do some housekeeping after upgrades and be sure that there are no old libraries which could disturb.

There is no way for the source to check which or how many of a given library is being used, is there?   I think multiple copies of the LCD library have caused problems before.

thinkyhead, 2016-04-14:
> which or how many of a given library is being used?

Only if the library includes a define for its version. All libraries should, but probably there are a lot that don't.

fbarcenas, 2016-04-19:
With my last tests i found than the version of arduino always is correct for me:
(https://github.com/MarlinFirmware/Marlin/issues/3368)

jbrazio, 2016-05-04:
This has been addressed with PR #3462

closed 2016-05-04 by jbrazio