MarlinFirmware/Marlin #3468: Travis tests may not reflect real world

jbrazio, 2016-04-11:
I see that Travis copies some kind of custom `Arduino-core` before compiling the source code, does this really reflect a real world case ? I take myself as an example, when I compile I use the stock `Arduino-core` I never use the custom one being shipped with Marlin, I can't imagine anyone doing the same thing as Travis is doing to compile Marlin.

This brings me to a question, why are we shipping a custom `Arduino-core` ?

thinkyhead, 2016-04-12:
It is an oddity. For my part, I only rely on Travis CI to catch my typos and make sure the code actually compiles. Apart from that, it's not used for any heavy diagnosis. But maybe it could be.

jbrazio, 2016-04-14:
@thinkyhead but do you know why we use a custom core ? I would start by removing this custom core from the Travis build as only his action will bring us closer to the real world test condition.

thinkyhead, 2016-04-15:
I don't know the reasons why there is a custom Arduino-core, unless it is for some historical reason. We could look at the early commits where it was being worked on more.

thinkyhead, 2016-04-15:
Note that these may not be "custom" cores that are included with Marlin. They may only be included as "the cores that Marlin is known to support" and may actually be identical to their origins. But we would have to look and see if they are in fact unaltered.

jbrazio, 2016-05-10:
Closes with merge of #3715

closed 2016-05-10 by jbrazio