MarlinFirmware/Marlin #3470: Marlin-RC5 x/y motion slowed when LCD unplugged.

edwilliams16, 2016-04-11:
I have a RIGIDBOT with the factory LCD.  On upgrading from their custom Marlin 1.0.0 to Marlin 1.1.0-RC5, I found that x/y travel motion was drastically slowed, whether commanded from Simplify3D or Pronterface.  Z-motion and homing were normal speed.
  It turned out this was associated with the LCD being unplugged.  At least with the old Rigidbot supplied firmware, I was subject to random reboots when the LCD was attached.  I'm hoping this behavior will go away with the latest firmware.  if it does, I guess this is barely a bug, since I will keep it plugged in.  
   However, it does suggest that the firmware should test for the presence of the LCD initially and not attempt to update it later if it isn't there  - which I'm assuming is the cause of the slowdown.

thinkyhead, 2016-04-12:
If you configure Marlin to use an LCD it always uses it, and there are no conditions to check if it's being unplugged and plugged back in while the machine is on. If your LCD is giving you problems, I suggest re-flashing the firmware with LCD support disabled, and then just don't use it.

edwilliams16, 2016-04-12:
Workaround is fine.  
   In the older version of Marlin unplugging the LCD had no effect - now it does. However, the good news is I may not need to unplug it!  I've gone several hours without a random reboot, so the later Marlin may well have fixed that problem.

thinkyhead, 2016-04-12:
@edwilliams16 We're fixing bugs left and right, so we might have fixed issues causing reboot or other such odd issues in the process without even knowing it. Of course, please thump us on the heads if you see any more crashing, rebooting, or other bizarre things!

closed 2016-05-04 by jbrazio