MarlinFirmware/Marlin #3483: Stuttering on curved movement (Pause between GCODE?)

snowzach, 2016-04-13:
So I've been doing some test prints on my Cartesian/RAMPS using RCBugFix and I haven't noticed any issues until today when I started printing something with curved surfaces. I've mostly been printing calibration cubes which seems to go fine. 

I've noticed today that it's stuttering really bad. I suspect it's in between GCODE's its got some sort of delay as I said when moving in a straight line it works perfectly. I tested both from an SDCARD and OctoPrint via USB. It's actually printing correctly too so it's not likely from missing steps.

jbrazio, 2016-04-13:
Hello @snowzach please have a look at #3129.

snowzach, 2016-04-13:
I'm not sure I agree with this being an issue with the slic3r. This is basically saying my slic3r is making code that's too hard. The code is more than capable of handling gcode that's lots of small movements, it just means not updating the LCD as often. I would rather have my prints faster and without tons of stuttering than an LCD that's perfectly updated in every loop.

I just put together a quick patch that actually works pretty well... I'm kinda git stupid so I'm not sure the best place to put it...

Basically, in Marlin_main.cpp we start measuring how many commands we processed in the last BUSY_CHECK_INTERVAL. If it's more than BUSY_COMMAND_PER_INTERVAL_COUNT we set a global flag that we're busy. If the busy flag is set, we only call lcd_update() every LCD_UPDATE_BUSY_INTERVAL milliseconds. 

```
// Top of Marlin_main.cpp
millis_t last_busy_measure_start_time = 0;
millis_t now = 0;
millis_t last_lcd_update = 0;
#define BUSY_CHECK_INTERVAL 500
#define BUSY_COMMAND_PER_INTERVAL_COUNT 3
#define LCD_UPDATE_BUSY_INTERVAL 200
bool busy = false;
uint8_t busy_command_count = 0;

//top of loop()
now = millis();
  if(now - last_busy_measure_start_time > BUSY_CHECK_INTERVAL) {
    if(busy_command_count > BUSY_COMMAND_PER_INTERVAL_COUNT) {
      busy = true;
      SERIAL_ECHOLN('BUSY');
    } else {
      busy = false;
    }
    last_busy_measure_start_time = now;
    busy_command_count = 0;
  }

// inside of idle() replacing lcd_update();
  if(busy) {
    if(now - last_lcd_update > LCD_UPDATE_BUSY_INTERVAL) {
      lcd_update();
      last_lcd_update = now;
    }
  } else {
    lcd_update();
    last_lcd_update = now;
  }
```

jbrazio, 2016-04-13:
I'm not saying it's the same issue, could be related. We reached the conclusion that in that case the error was in the file because printing with the LCD enabled on the config or not produced the same result.

If you "disable" the LCD and print the same gcode it works well ?

snowzach, 2016-04-13:
Yes, if I disable the LCD, it prints well. Sorry, I should have said that.

snowzach, 2016-04-13:
I tried updating the LCD every 400ms when it's busy and it works as well... Honestly, 200 it's hardly noticable. Maybe it's worth just setting the Graphic LCD to only update every 200ms

snowzach, 2016-04-13:
HAAA.. I fixed this even better with 2 lines of code... The LCD will get a little bit sluggish when it's doing all sorts of complex curves but it's still very usable. (at least for me)

In idle() inside Marlin_main.cpp update the lcd_update(); call to only call if the steppers have work to do or there's no commands in the queue.

```
  // If we're busy processing blocks (for the steppers) 
  // or there's no commands in the queue then update the LCD
  if(current_block != NULL || commands_in_queue == 0) {
    lcd_update();
  }
```

snowzach, 2016-04-13:
I'm going to close this and open a more specific case to my LCD.

closed 2016-04-13 by snowzach

paulusjacobus, 2016-04-13:
@snowzach many people encountered that stuttering behaviour and it's mostly related to the slicer software or printing at a high speed 150 mm/s. Try to print a tiny cylinder or a clog wheel and see if it seriously stutters or become worse to get confirmation that it is slicer related and try different settings in the slicer. I use Cura and so far I have a very little stuttering during the printing. Although i print at a very slow pace 40 mm/s on my mendel prusa it2.