MarlinFirmware/Marlin #3484: Not sure what is being accomplished with #define SDSUPPORT

dammitcoetzee, 2016-04-13:
I've been fighting with my firmware for the past day or so after upgrading to RC5. I copied my config and pins over like I always do, except this time my SD card was mysteriously not working. I'm running a rambo and reprapdiscount smartlcd controller

I dug through the code, tried out other RCs, etc. No avail. Until just moments ago when I was really really carefully reading the code in desperation. I finally spied under #define DOGLCD and above a huge block of other commented code, the object of my search: #define SDSUPPORT :|

I'm really not certain 
a) Why this is in such a weird place. If you want to be THAT granular about enabling features. (Even ones that people obviously want. Seriously, I don't think I've met a printer without SD card support.) Can we at least have it at the top of the category?

b) Why I wouldn't want to use the SD card soldered to the back of my display? I can't think of a reason. So, to me at least, it seems weird to have the edge case enabled by default. I used to enable my controller and get all its features. Now I have to know to go to two different places to enable the same board? Seems like a step back in usability to me. I think for cards that support it, it would make sense to have #define SDSUPPORT in their definition, and have a NOSDSUPPORT option to turn it off in config.h if desired. 

Anyway, despite my first world rage, I would like to add that the new firmware works much better than my old one. The code is much cleaner than it was, and I like the streamlined display! So thanks! :)

jbrazio, 2016-04-13:
Thank you for this report, we do not get users reporting their experiences/frustrations unless they are really unable to fix them.

> I copied my config and pins over like I always do

Why are you copying a custom `PINS` file, isn't your board by default included into Marlin ?

RCBugFix and also RC4, RC5 had a lot of configuration changes, we tried to warn the users for the most obvious ones by breaking the compilation but some stuff can go under the radar, it's always advisable to have a quick diff between a stock config and a "new current" config just to understand what changed.

> I finally spied under #define DOGLCD and above a huge block of other commented code, the object of my search: #define SDSUPPORT :|

I will make this more visible, we cannot assume every LCD user whats to have SDCard enabled despite most of the LCD boards nowadays comes with the SD slot builtin.

> Now I have to know to go to two different places to enable the same board

What exactly do you mean ? You should be able to have your printer fully setup by adjusting just two files, `Configuraiton.h` and `Configuration_adv.h`.

thinkyhead, 2016-04-14:
It used to be (See version 1.0.2) that `SDSUPPORT` was automatically enabled by every LCD that had one. At the same time there was an `SDSUPPORT` option in `Configuration.h` but whether or not you enabled or disabled it didn't matter. If you wanted to disable `SDSUPPORT`, well you couldn't.

So there was the question as to whether we should add a new option `DISABLE_SDSUPPORT` or make `SDSUPPORT` user-configurable, and we ended up opting for a reduction.

We tried out `DISABLE_SDSUPPORT` at first, but it wasn't working so well. Some code was being improperly compiled because `SDSUPPORT` was set before `DISABLE_SDSUPPORT` was able to override it. I don't think we would have the same problem with the current codebase.

Let me put together a PR with `DISABLE_SDSUPPORT` and see how it looks….

lrpirlet, 2016-04-14:
I would suggest to change in configuration.h
```C++
//#define SDSUPPORT // Enable SD Card Support in Hardware Console 
770                     // Changed behaviour! If you need SDSUPPORT uncomment  
```
to 
```C++
#define SDSUPPORT // Enable SD Card Support in Hardware Console 
770                     // Changed behaviour! If you do not need SDSUPPORT, do comment  
```
to achieve requested results...
or, did I fail to understand?

jbrazio, 2016-04-14:
> Let me put together a PR with DISABLE_SDSUPPORT and see how it looks

I would suggest to keep the option as it is.

> I would suggest to change in configuration.h #define SDSUPPORT

This may cause errors if the board has no SD support.

I would suggest the following change: Make really visible inside `Configuration.h` that the option `SDSUPPORT` exists and is disabled by default, if you look now at the file you'll see this option "hidden" in the middle of a "comment block" making it hard to find if you don't know what you're really looking for.

lrpirlet, 2016-04-14:
@jbrazio  Great idea.... What about associating with each and every option a unique string... for example **option that need user decision**
```
//#define SDSUPPORT // option that need user decision
                    // Commented will disable SD Card Support in Hardware Console
                    // Uncommented will enable SD Card Support in Hardware Console 
...
#define WHATEVER // option that need user decision
                 // Commented will stop people wondering
                 // Uncommented will make people search for the effect of that option  
```
Well, you get the idea, do not introduce WHATEVER option... :-)

jbrazio, 2016-04-14:
Some options belong sequenced together but yes you're right each options should have its own documentation block.

thinkyhead, 2016-04-14:
Perhaps when `ULTIPANEL` is enabled we should issue a warning:
```cpp
#if DISABLED(SDSUPPORT)
  #error Your controller has an SD Card slot! Enable SDSUPPORT to use it, or comment out this line to continue without it.
#endif
```

jbrazio, 2016-05-04:
Thank you for your interest making Marlin better and reporting this issue but this topic has been open for a long period of time without any further development. Marlin has been under heavy development for the past couple of months and moving to it's last mile to finish the RC cycle and release **Marlin v1.1.0**. We suggest you to try out the latest [RCBugfix](https://github.com/MarlinFirmware/Marlin/archive/RCBugFix.zip) branch and reopening this issue if required.

closed 2016-05-04 by jbrazio

kpdelucia, 2018-02-10:
2018 i am looking for the final solution in marlin 1.1.0  i have mks board and reprapdiscountsmartcontroller and have lost sd card funtionality.  (hictop 3dp-03) using dual extruder

Bob-the-Kuhn, 2018-02-13:
@kpdelucia - please open a new issue.  In the new issue please provide more information on the problem and what you've tried.

Before that try bugtfix-1.1.x.  Newer firmware may have solved the problem.