MarlinFirmware/Marlin #3485: REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER and lots of small GCODE moves cause stuttering

snowzach, 2016-04-13:
I ran into the same issue as here [#3129](https://github.com/MarlinFirmware/Marlin/issues/3129) but I reject the answer is that it's the slicers fault. The Marlin code is more than capable of handling the movements and I think it's more important that the steppers move quickly and smoothly vs the LCD get updated as rapidly in the case where there is a lot of movement going on.

So rather than bitch, I'll give you a solution... 

In idle() inside Marlin_main.cpp update the lcd_update(); call to only call if the steppers are busy working there's no commands in the queue. current_block is a pointer to what the steppers are working on via the interrupt routine and commands_in_queue are all the pending commands. This will ensure that if the steppers have run out of work and there's still commands in the queue, we'll skip updating the display this loop. 
```
// If we're busy processing blocks (for the steppers) 
// or there's no commands in the queue then update the LCD
if(current_block != NULL || commands_in_queue == 0) {
  lcd_update();
}
```
I have found that my display will get a tad sluggish when it's busy drawing a complex curve but it's still very usable. My steppers continue to run full speed during the curves.

Blue-Marlin, 2016-04-13:
Currently a display refresh of the info screen occurs about 1/second.
For no display the refresh needs about 0ms.
For a REPRAP_DISCOUNT_SMART_CONTROLLER a display refresh needs about 25ms
For a REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER a display refresh needs about 80ms.
For a VIKI2 a display refresh needs about 175ms.

In % this is about:
No display 1000-0 -> 100%, 0 -> 0%
REPRAP_DISCOUNT_SMART_CONTROLLER 1000-25 -> 100%, 25 -> 2.564%
REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER 1000-80 -> 100% 80 -> 8.695%
VIKI2 1000-175 -> 100%, 175 -> 21.212%

So what you get with omitting some screen updates is a tiny shift of the border case by up to 21%.

Increasing the minimal segment length in the slicer to a value equal or above
```
const unsigned int dropsegments = 5; //everything with less than this number of steps will be ignored as move and joined with the next movement
```
is much more efficient.
In the data we analysed in #3129 the most critical pars need ~4 lines of g-code to plan about 5 steps.
Even with `dropsegments = 1` not every g-code line produced a step. Sending lots of g-code, parsing that, partially planning that, to throw it away,most of the time - is really inefficient.

Based on data from: https://github.com/AnHardt/Marlin/pull/13#issue-126298038

Your idea will improve Marlin a bit, if you can live with the rare updates, but don't expect wonders.

snowzach, 2016-04-13:
I'll look at my slicer settings. All I can say now though is that without that code it stutters on every corner vs with it, it runs smooth as silk.

jbrazio, 2016-04-13:
@Blue-Marlin I saw some estimation from @thinkyhead and it was around 10Hz refresh rate for the LCD, you're speaking 1Hz.. did I miss interpreted him or we're not aligned ?

jbrazio, 2016-04-13:
@snowzach do you mind testing @AnHardt PR AnHardt/Marlin#13 to see if it improves your situation ?

Blue-Marlin, 2016-04-14:
@jbrazio 
Wat he wanted to say is `lcd_update()` is called about 10 times a second.But therein we find
```
    if (currentMenu == lcd_status_screen) {
      if (!lcd_status_update_delay) {
        lcdDrawUpdate = LCD_DRAW_UPDATE_CALL_REDRAW;
        lcd_status_update_delay = 10;   /* redraw the main screen every second. This is easier then trying keep track of all things that change on the screen */
      }
      else {
        lcd_status_update_delay--;
      }
    }
    if (lcdDrawUpdate) {
// Do the real display update here
    }
```
I did not test if this is still working (after the change to named states), but it should.

snowzach, 2016-04-14:
I do know that if I put in logic to make it so lcd_update is only called once every 200ms, it seems to fix my particular problem.

snowzach, 2016-04-14:
Okay, so I refined my original idea of calling lcd_update when there was sufficient work for the steppers and it's actually working how I intended now. I check that the block buffer is full or there are no commands in the queue and if so, update the lcd. The LCD is actually still pretty responsive. I think the upshot if this could be that for lcds with worse update intervals, you could potentially increase the block buffer. I'm not sure if there are any other drawbacks with this but so far it seems to be working quite well for me.
```
In idle(), replace lcd_update(); with: 

  // If the block buffer is full or there's no commands in the queue update the lcd
  if(movesplanned() >= BLOCK_BUFFER_SIZE-1 || commands_in_queue == 0) {
    lcd_update();
  }
```

snowzach, 2016-04-14:
Wow... So talk about an exotic one... So it turns out the issue here is that I had: #define ENCODER_PULSES_PER_STEP -1 which I used to reverse the scrolling when REVERSE_MENU_DIRECTION wasn't available.

Well, for whatever reason this will bypass the update 1 of every 10 times code of the LCD and force it to redraw every call to lcd_update() which causes the delay and stepper jitter problem I was experiencing. I switched those out and it's working well now. 

Feel a bit stupid now.. thanks for the help guys.. Closing this.

closed 2016-04-14 by snowzach

jbrazio, 2016-04-14:
Not stupid at all, really a nice catch !