MarlinFirmware/Marlin #3486: error: 'DEBUG_POS' was not declared in this scope

Roxy-3DPrintBoard, 2016-04-13:
There is a:
#define DEBUG_POS(PREFEX,VAR) do{ ...}  

later...  But the first reference to it in the Marlin_main.cpp file is above it....

Marlin_main.cpp:494:86: error: 'DEBUG_POS' was not declared in this scope

    if (DEBUGGING(LEVELING)) DEBUG_POS("sync_plan_position_delta", current_position);

jbrazio, 2016-04-13:
@Roxy-3DPrintBoard I can't reproduce the error on my side, which config are you using ?

Roxy-3DPrintBoard, 2016-04-14:
Well...  Actually... You don't even need to have the configuration parameters...  If you look where DEBUG_POS is used, it hasn't been defined yet.   

But I was using something close to:
~~~~
#define AUTO_BED_LEVELING_FEATURE // Delete the comment to enable (remove // at the start of the line)
#define DEBUG_LEVELING_FEATURE
#define Z_MIN_PROBE_REPEATABILITY_TEST  // If not commented out, Z Probe Repeatability test will be included if Auto Bed Leveling is Enabled.
~~~~

Just turning on DELTA or SCARA should be enough to cause it.

thinkyhead, 2016-04-14:
Sorry about that – more patches to print_xyz are coming too….

Roxy-3DPrintBoard, 2016-04-14:
That code is in the M114 handler.   I just commented out the problem and will continue my debug session later today.     

I already turned off the Z_PROBE_ALLEN_KEY stuff.   And I'm going to re-read all the text talking about Z_MIN Probes...  Maybe I had some mental block keeping me from understanding the right things to turn on.

jbrazio, 2016-04-14:
You could move the `#define DEBUG_POS(PREFEX,VAR) do{ ...}` into Marlin.h inside a `#ifdef` block. But @thinkyhead has greater plans for the overall solution so I propose to wait for his final PR instead of messing with a temporary solution.

thinkyhead, 2016-04-15:
Ok, #3504 adds debugging support for `vector_3` and earlier commits moved definition of `DEBUG_POS` closer to the top of the source.

closed 2016-04-15 by thinkyhead

Roxy-3DPrintBoard, 2016-04-15:
Why does the HAS_Z_PROBE macro act sick on my machine?

thinkyhead, 2016-04-15:
@Roxy-3DPrintBoard Is there a compile problem or does it act up when you run Marlin? What is the `Z_MIN_PROBE_PIN` value in your board's pins file? What kind of probe are you using? And what probe options do you have enabled in the config?