MarlinFirmware/Marlin #3489: MAXTEMP BED triggered when requesting Preheat ABS All

snowzach, 2016-04-14:
I'm re-opening this issue as it wasn't resolved. 
Cartesian, RAMPS, 2 Extruders. 
If I select Preheat ABS Both it immediately triggers MAXTEMP BED. I did some digging and it almost looks to me like there is a memory overrun somewhere.

In the ISR for temp I was watching the values over serial for current_temperature_bed_raw and bed_maxttemp_raw and they seem completely correct at startup as well as when the ISR runs.

As near as I can see, bed_maxttemp_raw is only ever modified at the end of the tp_init() function. I added a serial debug message there as well so I could monitor when bed_maxttemp_raw is modified. It modifies it a bit at startup and everything looks normal.

When I then select Preheat ABS Both the value for bed_maxttemp_raw immediately jumps however the debug in tp_init is never hit so I'm not sure where else it could be modified unless there something not behaving itself.

thinkyhead, 2016-04-14:
Sorry about that. The problem is apparently with the `lcd_preheat_pla0123` and `lcd_preheat_abs0123` functions. If you replace them with these patched versions it should fix the issue:
```cpp
  void lcd_preheat_pla0123() {
    #if EXTRUDERS > 1
      setTargetHotend0(plaPreheatHotendTemp);
      setTargetHotend1(plaPreheatHotendTemp);
      setTargetHotend2(plaPreheatHotendTemp);
    #endif
    _lcd_preheat(EXTRUDERS - 1, plaPreheatHotendTemp, plaPreheatHPBTemp, plaPreheatFanSpeed);
  }
  void lcd_preheat_abs0123() {
    #if EXTRUDERS > 1
      setTargetHotend0(absPreheatHotendTemp);
      setTargetHotend1(absPreheatHotendTemp);
      setTargetHotend2(absPreheatHotendTemp);
    #endif
    _lcd_preheat(EXTRUDERS - 1, absPreheatHotendTemp, absPreheatHPBTemp, absPreheatFanSpeed);
  }
```
I will submit a PR right away with these changes!

snowzach, 2016-04-15:
Ahh... I was just sitting down to hunt this bug down and you got it solved already. Well Done. 

This firmware is now my daily driver.. First real print on my new printer using RCBugFix (Frog)[http://imgur.com/JrHHDmQ]

closed 2016-04-15 by snowzach

thinkyhead, 2016-04-15:
@snowzach Now is a good time to be using 1.1.0 `RCBugFix`. The code is getting to a very stable and bug-free point now.