MarlinFirmware/Marlin #3494: active low endstop

ruggb, 2016-04-14:
I have always questioned this.
"// Mechanical endstop with COM to ground and NC to Signal uses "false" here (most common setup)."
I'm an engineer and the approved method for external signal input is active low (which would be "NO to Signal". My Ramps/Arduino boards implement active low. I have no idea what the rest of the boards use but I find it hard to believe that this is "(most common setup)".
If it is, that is fine. If not, changing these settings to a default of TRUE would be good.
For an explanation, referring to these as ACTIVE LOW or ACTIVE HIGH may be less confusing than "with COM to ground and NC to Signal" but that is an engineer speaking.

jbrazio, 2016-04-14:
I would say from the example configurations that both switch options are fairly common. But I do agree we could have better comment description for this section.

thinkyhead, 2016-04-15:
Why "normally closed" switches? Well, to quote random guy from reprap forum post:
> This is really a holdover from larger scale industrial machines common control design practice (and makes sense at this scale as well). If a limit switch fails on a punch press that handles sheets of 12 ga steel, someone can get seriously hurt (not to mention property damage) if the machine does not default to a halt state when a sensor defect occurs.

If a Normally-Open switch fails it will not give any trigger, whereas Normally-Closed will obviously fail if the wire breaks. NC switches should also be more immune to false triggers by induction.

closed 2016-04-17 by jbrazio