MarlinFirmware/Marlin #3497: Problem with HAS_Z_PROBE macro

Roxy-3DPrintBoard, 2016-04-14:
I'm fighting a horrible problem that doesn't make sense to me.   I have the RCBugFix repository at this location if I can talk anybody into taking a look at it:

https://github.com/Roxy-3DPrintBoard/Marlin

I can't get my current configuration to compile because the SanityCheck.h is blowing up saying I need a Z-Probe.  But I do have a Z-Probe defined.   I've traced it down to SanityCheck.h thinking there is no pin defined for Z_PROBE_MIN_PIN.   But if you look at the pins_UltraMaker.h file, I have it #defined to 30.    This is the current pins configuration but I've also tried it with Z_MIN_PIN set to 30 and totally undefined.
~~~~
//#define Z_MIN_PIN 30
#define Z_MIN_PIN -1		// Changed by Roxy
#define Z_MIN_PROBE_PIN 30	// Changed by Roxy
~~~~

I've added a couple of lines to SanityCheck.h to prove it is the HAS_Z_PROBE macro that is blowing up:

~~~~
#if !HAS_Z_PROBE
  #error Huh???  No Z-Probe?  How can that be?
#endif
~~~~
And I get this set of error messages.  I've attached a .ZIP file of the source files if anybody is willing to help me figure this out!   Thanks!
~~~~
Arduino: 1.7.9 (Windows 7), Board: "Arduino Mega or Mega 2560, ATmega2560 (Mega 2560)"
.
Using library Wire in folder: C:\Program Files (x86)\Arduino\hardware\arduino\avr\libraries\Wire 
Using library LiquidCrystal in folder: C:\Program Files (x86)\Arduino\libraries\LiquidCrystal 
Using library SPI in folder: C:\Program Files (x86)\Arduino\hardware\arduino\avr\libraries\SPI 
.
C:\Program Files (x86)\Arduino/hardware/tools/avr/bin/avr-g++ -c -g -Os -w -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10709 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR -IC:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino -IC:\Program Files (x86)\Arduino\hardware\arduino\avr\variants\mega -IC:\Program Files (x86)\Arduino\hardware\arduino\avr\libraries\Wire -IC:\Program Files (x86)\Arduino\libraries\LiquidCrystal\src -IC:\Program Files (x86)\Arduino\hardware\arduino\avr\libraries\SPI C:\Users\Roxy\AppData\Local\Temp\build8161014942809678390.tmp\blinkm.cpp -o C:\Users\Roxy\AppData\Local\Temp\build8161014942809678390.tmp\blinkm.cpp.o 
In file included from C:\Users\Roxy\AppData\Local\Temp\build8161014942809678390.tmp\Configuration_adv.h:689:0,
                 from C:\Users\Roxy\AppData\Local\Temp\build8161014942809678390.tmp\Configuration.h:1032,
                 from C:\Users\Roxy\AppData\Local\Temp\build8161014942809678390.tmp\Marlin.h:44,
                 from C:\Users\Roxy\AppData\Local\Temp\build8161014942809678390.tmp\blinkm.cpp:28:
C:\Users\Roxy\AppData\Local\Temp\build8161014942809678390.tmp\SanityCheck.h:176:4: error: #error A probe needs a pin! [Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN || HAS_Z_PROBE]
   #error A probe needs a pin! [Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN || HAS_Z_PROBE]
    ^
C:\Users\Roxy\AppData\Local\Temp\build8161014942809678390.tmp\SanityCheck.h:188:4: error: #error Huh??? No Z-Probe? How can that be?
   #error Huh???  No Z-Probe?  How can that be?
    ^
C:\Users\Roxy\AppData\Local\Temp\build8161014942809678390.tmp\SanityCheck.h:215:10: error: #error You must have a Z_MIN or Z_PROBE endstop to enable Z_MIN_PROBE_REPEATABILITY_TEST.
         #error You must have a Z_MIN or Z_PROBE endstop to enable Z_MIN_PROBE_REPEATABILITY_TEST.
          ^
Error compiling.
~~~~

jbrazio, 2016-04-14:
If I `#define Z_MIN_PROBE_ENDSTOP` it allows compilation with a lot of warnings.. ss you are doing this:
```
#define Z_MIN_PIN -1		// Changed by Roxy
#define Z_MIN_PROBE_PIN 30	// Changed by Roxy
```

```
// Enable Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN to use the Z_MIN_PIN for your Z_MIN_PROBE.
// The Z_MIN_PIN will then be used for both Z-homing and probing.
#define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN
```

From the comment it seems contradicting with the configuration file for your board as you are `#define Z_MIN_PIN -1` thus disabling it ?

Roxy-3DPrintBoard, 2016-04-15:
Well, like I said in a different thread.  Somebody spent a lot of time and effort trying to help people do the right thing.  But from my perspective, a Z_MIN endstop is different than a Z-Probe.   You can use a Z-Probe as a Z_MIN endstop just because of how the code evolved.  But we need a different way to communicate how to set these Z-Probes up because even I'm having trouble with it.

thinkyhead, 2016-04-15:
Although there are some exceptions that need to be cleaned up, I think that pins should always be defined if they exist on the board. In other words there should be no place in a `pins_MYBOARD.h` file where it says:
```cpp
#if ENABLED(SOME_OPTION)
  #define THAT_PIN 22
#else 
  #define THAT_PIN -1
#endif
```
The pins we usually find silk-screened on the board as "X-MIN", "X-MAX", etc., are the pins that Marlin calls `X_MIN_PIN`, `X_MAX_PIN`, etc. Some boards define a separate pin `Z_MIN_PROBE_PIN` in addition to the endstop pins. On the RAMPS boards this pin is a different pin from the endstop pins, but configurations or pins files can define it to coincide with any of the endstop switches known to be unused, or may share a pin. For instance for the Azteeg X3 Pro, `Z_MIN_PROBE_PIN` as the same as `Z_MIN_PIN`.

Whether certain code gets compiled now depends less on pins being set != -1, and more on options being set. Even so, currently this code in `pins.h` undefines `Z_MIN_PROBE_PIN` as an extra measure.
```cpp
#if ENABLED(DISABLE_Z_MIN_PROBE_ENDSTOP) || DISABLED(Z_MIN_PROBE_ENDSTOP) // Allow code to compile regardless of Z_MIN_PROBE_ENDSTOP setting.
  #undef Z_MIN_PROBE_PIN
  #define Z_MIN_PROBE_PIN    -1
#endif
```
I agree the number of probe options is becoming large and a little confusing. But I think this multiplication of granular options is necessary at first. Once we have a good set of known characteristics, we can group them in a similar way that we group LCD options now. And later on we'll be able to have a single setting like:
```cpp
#define Z_PROBE_TYPE MECHANICAL_PROBE
#define Z_PROBE_PLUG ZMAX
```
Then a setting like this one that we currently have users set manually…
```cpp
#define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN
```
…can be more automated…
```cpp
#if Z_PROBE_PLUG == Z_MIN && Z_ENDSTOP_PLUG == Z_MIN
  #define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN
#endif
```
…and stuff like that.

Roxy-3DPrintBoard, 2016-04-15:
OK...  Your reply helped explain things and got me compiling clean.   

@jbrazio We definitely need to get better, easier to understand documentation in place for setting up the Z-Probes.   And probably, that documentation also exists in the Configuration.h file because that is where a lot of other clues for setting stuff up exist.  Many people won't look anywhere else.

My problem was I thought if I turned something on, it was going to be used for that purpose.  I'm trying to get RC-5 moved over to a Delta printer and they home towards max.  My electronics board does have X-Min, Y-Min and Z-Min pins defined on it but I wasn't planning on using them.  So I didn't want them defined.

Another issue that threw me into the wrong mindset was this text in Configuration.h:
~~~~
// Enable Z_MIN_PROBE_ENDSTOP to use _both_ a Z Probe and a Z-min-endstop on the same machine.
// With this option the Z_MIN_PROBE_PIN will only be used for probing, never for homing.
~~~~
Somebody spent a lot of time and effort trying to help people.  But what I was thinking was "I'm not using a Z-Min Endstop.  So, obviously, this shouldn't be turned on.   And then at the end of this long comment it says:
~~~~
// Enable Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN to use the Z_MIN_PIN for your Z_MIN_PROBE.
// The Z_MIN_PIN will then be used for both Z-homing and probing.
#define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN
~~~~
Maybe some wording like this would have steered me down the right path:
~~~~
// Enable Z_MIN_PROBE_ENDSTOP to enable a Z-Probe that actually resides on the Z-min-endstop pin of the machine.
// With this option the Z_MIN_PROBE_PIN will only be used for probing.  The home location can optionally be probed.
~~~~

thinkyhead, 2016-04-16:
// **Specify here all the endstop connectors that are connected to _any endstop or probe_.**
    // Almost all printers will be using one per axis. Probes will use one or more of the
    // extra connectors. Leave undefined any used for non-endstop and non-probe purposes.
    #define USE_XMIN_PLUG
    #define USE_YMIN_PLUG
    #define USE_ZMIN_PLUG
    //#define USE_XMAX_PLUG
    //#define USE_YMAX_PLUG
    //#define USE_ZMAX_PLUG

thinkyhead, 2016-04-16:
> …some wording like this…

Good suggestions. Ideally I hope we can figure out a single setting to cover these Z probe and Z min cases in a cleaner way, or even automate them based on which pins are used, so we don't have to explain them so much.

closed 2016-04-16 by thinkyhead