MarlinFirmware/Marlin #3499: mirroring/inversion of part features

coalman321, 2016-04-14:
earlier today, i was attempting to print an arduino holder and when it was finished i noticed one thing. the tab part on the back was inverted causing the arduino to not fit in. i went through and checked the original file and the sliced code. they all had the correct location of the tab. what could be causing this?

Thanks

this is the post slicicng file...
[image]

coalman321, 2016-04-14:
Here is a picture of the printed version. 
[image]

Blue-Marlin, 2016-04-14:
> What could be causing this?

Having not a right handed printer coordinate system.

coalman321, 2016-04-14:
im not exactly sure what that means but ill elaborate a bit more just in case. this was made on a printrbot with a ramps 1.4 and marlin 1.1.0. is it a configuration issue? all other places i have read about mirroring, the device does not home correctly however this one does..
any help is appreciated,
thanks

thinkyhead, 2016-04-15:
It does look like the coordinates are inverted. Drop the STL here and the GCode output file and let's see what's happening there. What is your slicer?

coalman321, 2016-04-15:
attached is a zip file containing the stl and gcode files. they are both in correct orientation with an arduino layout .

thanks

[stlandgcode.zip](https://github.com/MarlinFirmware/Marlin/files/220793/stlandgcode.zip)

coalman321, 2016-04-15:
Oh and the slicer is sclic3r inside repetier host 
Thanks.

thinkyhead, 2016-04-16:
Until we figure out why your printer is doing this, try flipping the object in Repetier Host and re-slice it. Then it will print in the correct orientation.

I suggest you get the latest Marlin `RCBugFix` code and flash your board with that version. Then you can test one thing at a time:

- Homing to front-left of the bed?
- Printing small objects in the correct orientation and size?
- Acceleration, jerk, and speed limits produce good movement and surfaces?
- …and so on…

coalman321, 2016-04-16:
i can try this but as i am limited on time, it might take a while. the one thing i notice however is that it is just one axis. it is only doing this on the y axis and no others would this be the same situation?

thanks

emartinez167, 2016-04-16:
What if you invert the axis direction in the Configuration.h file?

#define INVERT_Y_DIR true

Regards,

Ernesto

Blue-Marlin, 2016-04-16:
If you have looked up what a "right handed" coordinate system is, and still think you have one, you likely have a machine with a bed moving in z. Don't look at the bed - that is not z. Z is the gap between bed and nozzle.

Setting `INVERT_Y_DIR  true` is a bad advice. This will ether change nothing, or crash your machine during the first athemt to home it.

Change  `INVERT_Y_DIR` to the opposit.
Make a y-max-endstop a min endstop or the other way around.
Change the `Y_HOME_DIR`

coalman321, 2016-04-16:
Ok after a closer and less scatterbrained attempt, one thing is evident. The y axis is fine but it is the x axis messing up. I attempted to change the home Dir and invert the y dir but it made the printer move out and not home. I'm going to try it with the x. 
Thanks

coalman321, 2016-04-16:
so a note on printrbot simples... they do min on z and y but max on x. it took a few hours (mostly due to endstops being weird) but it is finally fixed and working. im going to run 1 more print before i close this.

coalman321, 2016-04-16:
it was a success. thanks for your help.

closed 2016-04-16 by coalman321