MarlinFirmware/Marlin #3506: About time for RC6

thinkyhead, 2016-04-15:
Wow, we sure fixed a lot of bugs since RC5 and the code is definitely a lot better! One more release candidate should help tease out any bugs that might still be hiding in remote places. If there are no objections, I'll roll a new PR with an updated `README.md` tomorrow and we can put out another candidate for testing just in time for the weekend.

Between RC6 and the final release we can spend some time polishing up the documentation, especially on all the new probe and leveling configuration options.

paulusjacobus, 2016-04-15:
Thanks @thinkyhead I will hopefully have some time this weekend to test my
corexy and prusa mendel

christianh17, 2016-04-15:
HI! I would love testing a new RC. The last one (RCdev  some days after RC5) works without any problems for me (prusa 3 clone). Christian

garaguido, 2016-04-15:
Hi ,  I use now you RC5 ( first release 1-4-16) , but my man_bed_leveling dos not work on LCD
when I upload it  with Arduino IDE 1.6.8 .
When I upload RC5 with the verry old Arduino IDE 1.0.6 , the bed leveling from LCD is working , but I cannot print from LCD. 
Solution the new RC5 or RC6?

thinkyhead, 2016-04-16:
@garaguido Please give `RCBugFix` a try. We may have already fixed the bugs you're reporting.

garaguido, 2016-04-16:
Thanks , I will try it out and report you back about my findings.

garaguido, 2016-04-16:
Hi , I used your latest Marlin RCbugfix ( Latest commit a545290 6 hours ago)  and uploaded it with IDE 1.6.8 , here are the results :
- I can push the knob on "Prepare" and "Level bed" and then I turn it , notting happens , only the top row pixels from the text "Homing XYZ" is blinking (the text is not fully display missing top row pixels).
- The man bed level , still not working from my LCD , this was working when I uploaded with 1.0.6

thinkyhead, 2016-04-17:
@garaguido What kind of LCD controller are you using?

garaguido, 2016-04-17:
Yes , If you wish , I will post this as a new isseu.
My LCD controller is a ReprapDiscount Full Graphic Smart controller 128 x 64 from  Bigtree tech.com

scalez, 2016-04-19:
Can we try to get https://github.com/MarlinFirmware/Marlin/pull/3553 and https://github.com/MarlinFirmware/Marlin/pull/3554 into RC6 as well?

jbrazio, 2016-04-19:
Thumbs up for #3554.

thinkyhead, 2016-04-20:
Not #3553 at this time. #3554 is less "new" otherwise it wouldn't go in right now either.

jbrazio, 2016-05-04:
@thinkyhead I propose to keep track of what we want to include on RC6 using the [1.1.0 milestone](https://github.com/MarlinFirmware/Marlin/milestones/1.1.0).

closed 2016-05-24 by jbrazio