MarlinFirmware/Marlin #3507: Marlin had the function of the serial number, is used to identify the printer

liangruiben, 2016-04-15:
marlin有没有序列号的功能，用于识别打印机
Marlin had the function of the serial number, is used to identify the printer
当我有很多很多台打印机的时候，靠端口号来区分是不可能的事情了，因为端口号有可能更改的
When I have a lot of a lot of printer, by the port number to distinguish is impossible, because the port number may change

jbrazio, 2016-04-15:
You have the following on `Configuration.h`:
```
// Define this to set a unique identifier for this printer, (Used by some programs to differentiate between machines)
// You can use an online service to generate a random UUID. (eg http://www.uuidgenerator.net/version4)
//#define MACHINE_UUID "00000000-0000-0000-0000-000000000000"
```

When set it should be visible on serial output when sending the G-Code `M115`.

liangruiben, 2016-04-15:
谢谢，我准备开发服务软件，一个开源的，集中管理3D打印机，在搜集需要用到的固件支持
Thank you, I'm ready to develop service software, an open source, centralized management of 3d printers, collecting need firmware support

liangruiben, 2016-04-15:
可以在LCD上修改吗？
Can be modified on the LCD?

否则就需要每台打印机的固件都要不同
Otherwise we need every printer firmware should be different

jbrazio, 2016-04-15:
`MACHINE_UUID` must be set at compile time and currently cannot be changed. I believe we can accept this a Development Request for MarlinDev if enough interest is shown by the community for such feature.

thinkyhead, 2016-04-16:
@liangruiben All of these add-ons are possible. They are trivial, even. So why not fork Marlin and find a programmer in your local area to help you customize it as you need? The main Marlin project has other priorities right now. That is why we ask you to make this a Feature Request on the MarlinDev Issue Queue. We will get to it… someday. If you need features right now, you will have to hire help. Please pay them well for their time!

liangruiben, 2016-04-16:
marlindev 与marlin有什么不同，我是否应该使用marlindev
What's the difference between marlindev with marlin, marlindev should I use

thinkyhead, 2016-04-17:
MarlinDev is in a holding pattern, but it is the place where we will be implementing new features in Marlin 1.2.0 and beyond. It is on hold until we finish 1.1.0 – which is coming soon.

Roxy-3DPrintBoard, 2016-04-17:
Or stated differently...  MarlinFirmware/Marlin is where the stable and most bug free releases are.

MarlinFirmware/MarlinDev is where the evolution of Marlin happens.

closed 2016-04-23 by jbrazio