MarlinFirmware/Marlin #3508: Configuration option

sarf2k4, 2016-04-15:
I would like to suggest to add another configuration option. There will be an option to use simple configuration or the conventional.

if the simpler configuration were opted, those in the conventional either be overriden or not included or partially. The idea here is to let us who has been testing several things on several releases, we got to scroll a lot to check on specific things whether or not it matches. For example specifying `steps_per_mm` on the simple config would override the default values on default config.h. Same goes for enabling some options there

Furthermore the simple config shouldn't have much of a comments, just a pure config lines that will override the default

I've seen the `marlindev` branch last year somewhere july to november, the configuration there are profile based and there were no guidelines on setting up the profile based config file. 

The description might be almost the same idea as profile based, but it's actually not, in fact, this idea is to speed up the configuration process especially for testers whenever new RC has been released

Just an idea :D

jbrazio, 2016-04-15:
This concept of *inheritance* is implemented at MarlinDev.

sarf2k4, 2016-04-15:
I got confused on the marlindev configs

jbrazio, 2016-04-15:
I agree.. the concept is there but it's not yet polished.

sarf2k4, 2016-04-15:
That's the main reason I don't go for marlindev last time due to this concept. It is good but lack of proper documentation. 

So I was thinking that it might be a good idea if there are extra config file that is much simpler that can be permanent and those variables in defining them will be partial for example if I left the `x_endstop_pos -1` unchanged in the `default config.h` file and not defined in the `new simple config` file, the one in the `default` will be loaded

thinkyhead, 2016-04-16:
@sarf2k4 @jbrazio I vote for human usable configs (even as they currently exist) over any systematic inheritance system that splits configs up into lots of chunks. Even the companies that fork and use Marlin for their own firmware will tend to opt for the simpler configurations because they just have one guy assigned to the job, and there's no added value to adopting some brand new and complex configuration system when you can just edit one file and go home for the day.

jbrazio, 2016-04-16:
@thinkyhead you're right also !

I could suggest something like moving the existing `Configuration.h` and `Configuration_adv.h` to `DefaultConfiguration.h` and `DefaultConfiguration_adv.h`.

All options not found in the [new] `Configuration.h` and `Configuration_adv.h` would be defaulted to the ones set on the default files. If a user would like to set only PLA settings, their `Configuration.h` file could be as simple as:
```
#define PLA_PREHEAT_HOTEND_TEMP 180
#define PLA_PREHEAT_HPB_TEMP 70
#define PLA_PREHEAT_FAN_SPEED 0   // Insert Value between 0 and 255
```

CONSULitAS, 2016-04-16:
@thinkyhead @jbrazio 
You are both right, but:

Think of my situation:
* I take the default configuration and make a K8200 config.
  * Every time i do this, i check the deltas and correct my config.
* But my K8200 has modifications (like many of them, there are vendor kits for Z axis, extruder, ...).
  * So i change it again for my printer.

It is a mess!
* K8200 changes 46 lines in `Configuration.h` of nearly 900 and 14 in `Configuration_adv.h` of about 600 (comments not counted).
* For my own printer i have to change total 7 lines of this.
* To manage this i add `#include "Configuration_User.h` at the end of `Configuration_adv.h` with
```
// Jochens Spezial-Config

#ifndef CONFIGURATION_USER_H
#define CONFIGURATION_USER_H

#undef X_MAX_POS
#undef Y_MAX_POS
#undef Z_MAX_POS
#define X_MAX_POS 205
#define Y_MAX_POS 205
#define Z_MAX_POS 205

#undef LANGUAGE_INCLUDE
#define LANGUAGE_INCLUDE GENERATE_LANGUAGE_INCLUDE(de)

// TEMP
#undef  EXTRUDER_0_AUTO_FAN_PIN
#undef  EXTRUDER_AUTO_FAN_SPEED
#undef  EXTRUDER_AUTO_FAN_TEMPERATURE
#define EXTRUDER_0_AUTO_FAN_PIN 8
#define EXTRUDER_AUTO_FAN_TEMPERATURE 100 // should be 50 -> hack for problem with thermistor
#define EXTRUDER_AUTO_FAN_SPEED   255  // default 255 == full speed

#endif //CONFIGURATION_USER_H
```

I think the situation is similar for most printers.

So what i would propose is a
* `Configuration_default.h` like @jbrazio 
* `Configuration_machine.h` e.g. `Configuration_K8200.h` with only the changed values
* `Configuration.h` for the user which includes the machine and default and is place where the user changes his personal preferences

This will solve a lot of problems with the "example" configurations, too. I would like to call them maintained configurations (if they are).

jbrazio, 2016-04-16:
Agreed with the principle. But if we have a look at the dev source this is more or less what someone tried to do, the issues is:

1. Not all legacy configs were migrated, there is transitional support/confusion/mess on the tree.
2. They tried to split between vendor configs and something else.

I do like the "vendor idea", in fact what we currently call of "example configs" should be "vendor" which are well know configurations for the most basic kit we can find on the market for that particular printer model.

But we have to keep it simple.

thinkyhead, 2016-04-17:
I've got Javascript that can parse config files, so what is needed is for me to build a web page where you paste your old config on the left, and you get the new config on the right, and you're all done.

closed 2016-04-23 by jbrazio