MarlinFirmware/Marlin #3509: Picture files for status screen variants

tkurbad, 2016-04-15:
Hi,

I've got a little "feature" request:

Could someone please check in or point me to a source of the original bitmap files the status display C macros in dogm_bitmaps.h were generated from.

I want to change the status bitmap in my fork of Marlin, but I am having a hard time doing so. So, a "real" picture format (*.png, *.jpg, *.xbm, ...) of the graphical LCD elements would be really appreciated.

Or is there a way to convert the C macros back to a picture format?

Thanks in advance!

Best,
Torsten

Blue-Marlin, 2016-04-15:
https://github.com/MarlinFirmware/Marlin/blob/RC/Marlin/dogm_bitmaps.h#L24
https://github.com/AnHardt/Marlin/tree/lcd-speedup/Marlin/graphics
https://github.com/MarlinFirmware/Marlin/blob/RC/Marlin/fonts/README.fonts
https://github.com/MarlinFirmware/Marlin/blob/RC/Marlin/fonts/make_fonts.bat
https://github.com/MarlinFirmware/Marlin/wiki/LCD-Language

tkurbad, 2016-04-16:
> https://github.com/AnHardt/Marlin/tree/lcd-speedup/Marlin/graphics

Something like that

CONSULitAS, 2016-04-16:
@tkurbad 
Is your request answered successfully? If yes, please close this issue for housekeeping. Thanks in advance.

tkurbad, 2016-04-17:
Yes, I could accomplish what I was up to. Thanks!

closed 2016-04-17 by tkurbad