MarlinFirmware/Marlin #3510: M117 after leveling bed does not work anymore (RCBugFix)

Alex9779, 2016-04-15:
I am on RCBugFix of last night with a RepRap Discount LCD.
When doing the bed leveling (some lines are still missing there but that's already covered in another issue) and then doing a print M117 command in the code do not print on the LCD.
The LCD still says "Leveling finished" or what ever the message after leveling is.
After a reboot and doing nothing on the LCD it works though.
I have seen this issue too when doing something else in the menu but can't remember exactly what but maybe the first descriptions helps to track it down?

thinkyhead, 2016-04-16:
@Alex9779 What happens if you put `M400` right before `M117`?

thinkyhead, 2016-04-16:
Here's a solution for you. Change the line:
```cpp
LCD_ALERTMESSAGEPGM(MSG_LEVEL_BED_DONE);
```
…to…
```cpp
LCD_MESSAGEPGM(MSG_LEVEL_BED_DONE);
```

Alex9779, 2016-04-16:
So due to the merge I don't have to fix myself, right? And this is closed, right?

jbrazio, 2016-04-16:
@Alex9779 you only have to download the latest RCBugfix, the patch is merged there.
Usually you should close this ticket as soon as you confirmed the we have solved your problem.

closed 2016-04-17 by thinkyhead

Alex9779, 2016-04-17:
Oh ok, didn't know your policy how to handle this...