MarlinFirmware/Marlin #3511: Moving axis after a print shows weird symbols after a print (RCBugFix)

Alex9779, 2016-04-15:
I am on RCBugFix of last night.
This is an issue I realised AFTER a print finished.
I went to the prepare menu to lower the Z axis but the value just showed weird signs like + and . and other stuff when turning the knob.
Seems to work normally after a fresh boot though...
This issue appears on moving all axis...

thinkyhead, 2016-04-16:
Interesting! What version of Arduino are you using to build Marlin?

Alex9779, 2016-04-16:
Actually I realized this problem with several versions and had this issue testing RCs (4 and 5) and the RCBugFix for some weeks now.
Before this week I compiled on my Mac with 1.6.8
The version I have on my machine at the moment is compiled using 1.6.4 on a Linux virtual build machine, I adapted the build process you have setup for Travis. Generally it does the same, load Arduino, the libs but a little changed handling...

thinkyhead, 2016-04-17:
So the random characters are just in the edit-value portion of the display?

Alex9779, 2016-04-17:
Yes only there where the numbers should be are the random characters.

Alex9779, 2016-04-17:
I am going to test it again with a build of tonight and post some pictures...

thinkyhead, 2016-04-18:
Sorry, Alex, I forgot to remember… what kind of display are you using…?

thinkyhead, 2016-04-18:
Be sure to download the latest code for testing. We've been patching more bugs today, including a small buffer overflow issue.

Alex9779, 2016-04-18:
I have a "RepRapDiscount Full Graphic Smart Controller".
But I was not able to reproduce the issue anymore.
So going to close this... Thanks anyways, seems something fixed it.
I have an idea but I have to check first. Will reopen if it is necessary...

thinkyhead, 2016-04-19:
Thanks for the update. I just got hold of a Full Graphic Smart Controller and have been testing it. So far everything looks good for me also.

Alex9779, 2016-06-08:
This issue seems to be there.
[image]
This is RC6 with the RepRap Full Graphics Smart Controller after a one hour print.

jbrazio, 2016-06-08:
@Alex9779 Could you please try RCBugfix not RC6.
Thanks.

thinkyhead, 2016-06-08:
The E value is probably >= 1000. Perhaps we should be resetting it to zero when a print completes. Here's a workaround that you can implement in the `lcd_move_e` function:
```cpp
    float e = current_position[E_AXIS];
    while (e >= 1000.0) e -= 1000.0;
    lcd_implementation_drawedit(pos_label, ftostr31(e));
```

Alex9779, 2016-06-09:
Cool thanks...

closed 2016-06-09 by Alex9779