MarlinFirmware/Marlin #3512: Mesh bed leveling not lifting nozzle anymore when moving to next point? (RCBugFix)

Alex9779, 2016-04-15:
I am on RCBugFix of last night.
When doing a mesh bed leveling I realized that the nozzle is at 4mm above the bed for the first point.
Then turning the knob to adjust and click to move to next.
I remember that the RCBugFix as every firmware version I used before move the bed up to 4mm again before moving to the next point.
Today this did not happen.
Is this an intended new behavior? Did I miss some new configuration option?

epatel, 2016-04-15:
Maybe this change is causing it https://github.com/MarlinFirmware/Marlin/pull/3488

Alex9779, 2016-04-16:
So what I did, as far as I understand the discussion going on for that PR, I set MIN_Z_HEIGHT_FOR_HOMING in my configuration now.
Going to test when the current print finished...

Alex9779, 2016-04-16:
Doing what I wrote before solved the problem...

closed 2016-04-16 by Alex9779

thinkyhead, 2016-04-17:
Sorry about that. I wasn't intending to change the behavior. I have a PR ready to go shortly.

Incidentally, what are your `Z_MIN_POS` and/or `Z_HOME_POS` settings? Would you mind posting your configs?

I must admit (_mea culpa_) I haven't read the MBL documentation, so my settings might not be proper for MBL as I'm cleaning it up. Either way, I hope to make it work with "regular" settings so no special offsets are required.

thinkyhead, 2016-04-17:
Should be fixed now with the merge of #3524

Alex9779, 2016-04-17:
Cool many thanks!
I am a bit glad to see so that it is no only me who has problems seeing the "whole" thing whats going on.
That piece of code here is really hard to survey...