MarlinFirmware/Marlin #3517: Printing Cylinder Tube shape misalignment

guy2b1, 2016-04-16:
Hi guys,

Anyone knows the problem in kossel delta were after 10mm or more in z height, the nozzle starts dragging the part say more than 30mins have passed?  the parts result becomes misaligned especially when i print a cylinder or round tube shape parts, the edge sides no longer aligned that looks like it missed steps.  My stepper motors are calibrated properly when moved in certain distances aligned.  I am not sure what causing this.

thinkyhead, 2016-04-17:
Hi @guy2b1 —

- Do you see better or worse results when printing slower?
- What version were you using before `RCBugFix`?
- Have you done anything with the `M665` settings?

guy2b1, 2016-04-17:
Hi thinkyhead,

Same result either fast or slow in different random z height either below or above 10mm layers and in random time even before or more than 30 minutes of printing then misalignment starts and then the nozzle head starts dragging the topmost layer part that looks like its missing steps.  I already set the Lift Z to 4mm to avoid topmost layer part for dragging but the same result.

I tried using the stable Marlin 1.0.2.1 version and now RCBugFix versions but same misalignment and topmost nozzle head dragging result.  This only happens when I print circular tubular shapes when printing, but when I print a square hollow tube shape, the part came out perfect so I am assuming my steps per mm setting is correct, even dimensions is correct.

I am not sure about using M665 command as I am using Slic3r latest stable version and assuming that every gcode is created are automatically adjusted.

thinkyhead, 2016-04-17:
Are the grub screws on your pulleys nice and tight?

guy2b1, 2016-04-17:
Yes everything is tight, checked grub screws and belts already tight.

thinkyhead, 2016-04-17:
Do you have a cooling fan on your stepper drivers / electronics?

guy2b1, 2016-04-17:
Yes have 2 fans already running to cool the electronics.  I am using the Rumba Board.  I even change replace all DRV8825 drivers.  I even increase the trimpot to find the smooth spot.  But all the same problem misalignment dragging problrems.

thinkyhead, 2016-04-17:
What is your `DELTA_SEGMENTS_PER_SECOND` value?

guy2b1, 2016-04-17:
#define DELTA_SEGMENTS_PER_SECOND 200
  #define DELTA_DIAGONAL_ROD 293.0 
  #define DELTA_SMOOTH_ROD_OFFSET 213.1829
  #define DELTA_EFFECTOR_OFFSET 19.9
  #define DELTA_CARRIAGE_OFFSET 18.0
  #define DELTA_PRINTABLE_RADIUS 90

thinkyhead, 2016-04-17:
Do you see different results if you reduce `DELTA_SEGMENTS_PER_SECOND` to around `100`?

guy2b1, 2016-04-17:
Right now after printing the boat 3DBenchy, around 9 minutes in 7mm of layer height, the nozzle tip starts hitting dragging the topmost layer. It seems that instead of Z height going up, its going down to hit and drag the topmost layer.  Unfortunately still the same problem causing misalignment.

guy2b1, 2016-04-17:
i have done some reading and now I understand the DELTA SEGMENTS PER SECOND setting.  I will try to do some trial & error to find the sweet spot until I completely printed the 3DBenchy boat calibration part test.

thinkyhead, 2016-04-18:
The shortest path between two points on a delta is a curved line. So unless there's enough interpolation between points, and unless all the geometry settings are correct, there will be a natural dipping during moves. I'm not sure if this could be height-dependent under some circumstances. Someone who knows more about calibrating Delta machines will have to contribute.

As far as I know the maths for Delta are accurate in Marlin, and these algorithms haven't been altered in a long while. So I tend to think this is related to configuration rather than some logical bug.

guy2b1, 2016-04-18:
Shown in the pictures below this was printed around 1 hour from 23mm of layer height, then misalignment dragging starts then continues to print flat aligned again for the rest of 3mm layer height.  I was using the stable Marlin 1.0.2.1 during that time with the same misalignmend hotend tip dragging part problem.
[image]
[image]
Later I will try to experiment changing the DELTA SEGMENTS PER SECOND until I successfully print the boat calibration 3DBenchy stl file.  I hope this problem can be solved, as I never successfully 3d printed any part in my new built delta kossel printer.

thinkyhead, 2016-04-18:
I guess it's also important to have `SLOWDOWN` disabled on delta machines.

thinkyhead, 2016-04-18:
When layers shift like that it's usually because one or more of the steppers has fallen out of alignment, and this can happen for various reasons. For example, if you give too little or too much voltage to the steppers. But in your case it may only be falling out of alignment because the nozzle is dragging on the part. And that may be occurring due to bowl-shaped movements (bad delta calibration), or due to over-extrusion (too many E steps), or due to too few XYZ steps. You should also test to see how easy it is to intentionally knock the nozzle out of alignment. It shouldn't be very easy.

If you see exactly the same results every time, then I would suspect the firmware, and we can try to isolate the exact cause. But if the layer shift occurs at a different point each time, then it begins to look more like a hardware issue.

guy2b1, 2016-04-18:
I already checked all possible mechanical issues such as loose belts, grub screws, and intentionally forcing the motors to stop sliding up & down making it to miss steps, but its strong enough and smooth enough to slide up and down.  I even tried holding shaking the extruder and it is stiff bolted.  As I already change the firmware to RCBugFix version, yesterday I am printing the boat calibration 3DBenchy, in a random height layers say around 5mm or so it starts dragging the topmost part then misalignment starts everytime I attempted to do the 3d prints of this boat 3DBenchy.  I noticed that the dragging is happening in different heights not in a fixed layer height.

If adjusting the DELTA SEGMENTS PER SECOND to find the sweet spot is not successfull, I am thinking of switching electronic boards as I am currently using Rumba board, or otherwise switching to a 32bit such as Duet.  Or maybe order Ramps 1.4 to find out if the board has issues.  But the square  calibration shapes whether box or hallow, I can print it successfully without problems.

guy2b1, 2016-04-18:
As I went home, change DELTA SEGMENTS PER SECOND to 160 that came to my mind, then upload the firmware.  I immediately print the boat 3DBenchy in one go and was surprised, after printing for 1 hour and 20 minutes watching it, the boat came out perfectly complete without dragging and missing steps problem, Yahoo!!!  But theres a downside, I noticed that one of my stepper motors or probably my Y stepper motor has very weird loud sound,  I thought its going to miss steps while printing it  sounded like a scratching friction between 2 metals  i guess.  Vibrations are definitely present.  I am assuming my DRV 8825 trimpot was set to high voltage, will adjust it tomorrow and try to attempt a 2nd print test.  No mechanical or hardware adjustments made.

thinkyhead, 2016-04-19:
Voltage issues would certainly be a possibility here. Perhaps this issue is being exposed due to recent changes in acceleration behavior. We've got a topic where some are saying that for longer moves the acceleration seems to be instant. If you happen to notice any especially jerky movement, that may be contributing. But really, I have yet to reproduce that behavior, so I'm just speculating.

guy2b1, 2016-04-19:
Yes I guess its the voltage issue im having before I upgrade to RCBugFix as I remember I keep adjusting  the trimpot increasing the voltage troubleshooting guessing during that time something causing the misalignment and hotend tip dragging.   Now I downgraded to stable version Marlin 1.0.2.1 and looks working ok so far as I tried printing this vase tonight to observe the circular motion if it will cause to misalignment.  Thank you very much thinkyhead for all the help :)  Changing the Delta Segments Per Second also helps.

closed 2016-04-20 by thinkyhead