MarlinFirmware/Marlin #3518: Corexy - G29 not working in RC5

paulusjacobus, 2016-04-16:
@thinkyhead tested the lastest RCBugFix this weekend and found that the auto bed leveling matrix is not updated. Just shows its initial state after G29 is finished.
When printing the x coordinate shows garbage on the display OLED1306 i.e. ><--.0 and the carriage walks into the end stop. Y and Z coordinates show normal numeric values.
`>>>G29
SENDING:G29
gcode_G29 >>>
 > current_position: (154.00, 190.00, 0.00)
G29 Auto Bed Leveling
BEFORE matrix.set_to_identity > corrected_position: (154.00, 190.00, 0.00)
BEFORE matrix.set_to_identity > current_position: (154.00, 190.00, 0.00)
AFTER matrix.set_to_identity > uncorrected_position: (154.00, 190.00, 0.00)
sync_plan_position > current_position: (154.00, 190.00, 0.00)
setup_for_endstop_move > enable_endstops(true)
z_before = (before) 15.00
probe_pt >>>
> ProbeAction:1
 > current_position: (154.00, 190.00, 0.00)
Z Raise to z_before 15.00
> do_blocking_move_to_z 15.00
do_blocking_move_to: (154.00, 190.00, 15.00)
> do_blocking_move_to_xy 324.00, 30.00
do_blocking_move_to: (324.00, 30.00, 15.00)
> ProbeDeploy
deploy_z_probe > current_position: (324.00, 30.00, 15.00)
sync_plan_position > current_position: (324.00, 30.00, -2.56)
run_z_probe > current_position: (324.00, 30.00, -2.56)
<<< probe_pt
z_before = (between) 2.44
probe_pt >>>
> ProbeAction:0
 > current_position: (324.00, 30.00, -2.56)
Z Raise to z_before 2.44
> do_blocking_move_to_z 2.44
do_blocking_move_to: (324.00, 30.00, 2.44)
> do_blocking_move_to_xy 34.00, 30.00
do_blocking_move_to: (34.00, 30.00, 2.44)
sync_plan_position > current_position: (34.00, 30.00, 0.83)
run_z_probe > current_position: (34.00, 30.00, 0.83)
<<< probe_pt
z_before = (between) 5.83
probe_pt >>>
> ProbeAction:0
 > current_position: (34.00, 30.00, 0.83)
Z Raise to z_before 5.83
> do_blocking_move_to_z 5.83
do_blocking_move_to: (34.00, 30.00, 5.83)
> do_blocking_move_to_xy 34.00, 350.00
do_blocking_move_to: (34.00, 350.00, 5.83)
sync_plan_position > current_position: (34.00, 350.00, 0.75)
run_z_probe > current_position: (34.00, 350.00, 0.75)
<<< probe_pt
z_before = (between) 5.75
probe_pt >>>
> ProbeAction:2
 > current_position: (34.00, 350.00, 0.75)
Z Raise to z_before 5.75
> do_blocking_move_to_z 5.75
do_blocking_move_to: (34.00, 350.00, 5.75)
> do_blocking_move_to_xy 324.00, 350.00
do_blocking_move_to: (324.00, 350.00, 5.75)
sync_plan_position > current_position: (324.00, 350.00, -0.76)
run_z_probe > current_position: (324.00, 350.00, -0.76)
> ProbeStow (stow_z_probe will do Z Raise)
stow_z_probe > current_position: (324.00, 350.00, -0.76)
<<< probe_pt
> probing complete > current_position: (324.00, 350.00, -0.76)
clean_up_after_endstop_move > ENDSTOPS_ONLY_FOR_HOMING > endstops_not_homing()
Eqn coefficients: a: -0.00844087 b: 0.00270020 d: 0.78237934
>>> set_bed_level_equation_lsq > uncorrected_position: (324.00, 350.00, -0.76)
>>> set_bed_level_equation_lsq > current_position: (324.00, 350.00, -0.76)
<<< set_bed_level_equation_lsq > current_position: (324.00, 350.00, -0.76)
sync_plan_position > current_position: (324.00, 350.00, -0.76)

Bed Level Correction Matrix:
+1.000000 +0.000000 +0.000000
+0.000000 +1.000000 +0.000000
+0.000000 +0.000000 +1.000000
> BEFORE apply_rotation_xyz > z_tmp  = -0.76
> BEFORE apply_rotation_xyz > real_z = -0.76
> AFTER apply_rotation_xyz > z_tmp  = -0.76
sync_plan_position > current_position: (324.00, 350.00, 0.00)
> corrected Z in G29 > current_position: (324.00, 350.00, 0.00)
do_blocking_move_to: (324.00, 350.00, 15.00)
<<< gcode_G29`

thinkyhead, 2016-04-17:
Three backticks for multi-line quotes. See the "Styling with Markdown is supported" link.

If you're seeing this in RC5, sorry about that. This issue is now fixed in `RCBugFix`.

If you're actually using `RCBugFix`, sorry, this is supposed to be fixed in `RCBugFix`.

paulusjacobus, 2016-04-17:
I actually downloaded the RCBugFix release on Sat 17th which displays on my
printer display as RC5 hence the confusion. Do I something wrong by
selecting the release and download? Should the display say RCBugFix? So i
hope the fix was implemented and I will reload today and retest it, thanks.

thinkyhead, 2016-04-17:
The PR #3528 should fix this issue. Debug code was inserted in the wrong place, resetting the plane.

paulusjacobus, 2016-04-17:
Thanks @thinkyhead just tested the latest RCBugFix and it did fix the issues I saw. Haven't printed anything yet, just a dry run print with bed leveling.

I used the git hub tool to download the RcBugfix release which displays RC5 on my splash startup screen... is that normal? Close this now.

closed 2016-04-17 by paulusjacobus

thinkyhead, 2016-04-18:
The version in `RCBugFix` was `1.1.0-RC5` but I changed it to `1.1.0-RCBugFix` yesterday.