MarlinFirmware/Marlin #3520: Z axis Full speed without ramp

FalloutBe, 2016-04-16:
When moving the Z axis 10mm, I can clearly hear a rampup happening. When moving it 100mm, there is no ramp time at all?! There is also no ramp when homing the Z axis.

This happens since I updated from 1.1 RC3 to RC5.

thinkyhead, 2016-04-17:
How fast does your Z axis move at top speed?

FalloutBe, 2016-04-17:
40mm/s
I also tried this at 10mm/s and the same happened.

jbrazio, 2016-04-17:
Which printer do you have ?

FalloutBe, 2016-04-17:
A custom built one, ramps 1.4

jbrazio, 2016-04-17:
But do you suspect the robot is loosing steps after a 100mm move ? Or you're just noticing that no ramp up/down happens on longer moves ?

FalloutBe, 2016-04-17:
Well yes, it's not about loosing steps, I can just clearly hear and see that there is no ramp on the longer moves. Really strange

Pmant, 2016-04-17:
I'm not sure if I'm experiencing the same bug but I also can't hear/see any ramp when doing a g28.

jbrazio, 2016-04-17:
@thinkyhead did some modification to the acceleration code for RC4/5, let's wait for his feedback.

thinkyhead, 2016-04-18:
Try undoing this change if you think it might be involved: https://github.com/MarlinFirmware/Marlin/commit/614febaaf36fa349ebbc1f93ab61f669ea18bebe

Daemonentreiber, 2016-04-18:
I got a similar problem after updating from RC3 to RC5.
When homing the Z-axis just stutters and doesnt move.
Moving the Z-axis "manually" (with the lcd controller) works.
Same on RC4 and the latest RCBugfix. Switching back to RC3 and everything works fine.

thinkyhead, 2016-04-19:
@Daemonentreiber It sounds like your Z axis might be trying to move too fast. This could be due to a new bug, or it might be due to fixing a bug. What are your configuration settings for these options?
- `MANUAL_FEEDRATE`
- `HOMING_FEEDRATE`
- `DEFAULT_MAX_FEEDRATE`

thinkyhead, 2016-04-19:
If anyone feels like reverting the changes that I linked to above to see how it affects behavior, that would be helpful. I am not able to reproduce this issue with my hardware.

Daemonentreiber, 2016-04-19:
HOMING_FEEDRATE 3*60
DEFAULT_MAX_FEEDRATE 3
MANUAL_FEEDRATE 4*60

Undoing change [614feba](https://github.com/MarlinFirmware/Marlin/commit/614febaaf36fa349ebbc1f93ab61f669ea18bebe) seems to help.

thinkyhead, 2016-04-20:
@Daemonentreiber Thanks for the feedback. What if you only change the `unsigned long bsx` back to `long bsx`? Is that enough to get better behavior, or do you also need to revert the rest of https://github.com/MarlinFirmware/Marlin/commit/614febaaf36fa349ebbc1f93ab61f669ea18bebe to get it to behave? And what if the code is more like the code that Ultimaker used? Maybe it would help in those calculations to make sure everything is cast to `long` values so there's no doubt that there are enough bits to handle the large multiplication products.

Daemonentreiber, 2016-04-20:
Only changing the `unsigned long bsx` back to `long bsx` helps.

thinkyhead, 2016-04-21:
Aha! Very interesting… I wonder if @FalloutBe can also confirm…?

FalloutBe, 2016-04-21:
Thanks a lot for all your replies!
Maybe this weekend I'll have time to try this fix. I'll let you know then

thinkyhead, 2016-04-21:
I thought of another thing you can try, which might work just as well. Change this line in `planner.h`…
```cpp
  long steps[NUM_AXIS];                     // Step count along each axis
```
…to this instead…
```cpp
  unsigned long steps[NUM_AXIS];            // Step count along each axis
```

jbrazio, 2016-05-04:
Thank you for your interest making Marlin better and reporting this issue but this topic has been open for a long period of time without any further development. Marlin has been under heavy development for the past couple of months and moving to it's last mile to finish the RC cycle and release **Marlin v1.1.0**. We suggest you to try out the latest [RCBugfix](https://github.com/MarlinFirmware/Marlin/archive/RCBugFix.zip) branch and reopening this issue if required.

closed 2016-05-04 by jbrazio

FalloutBe, 2016-05-04:
I'm wondering,
I thought the issue was confirmed but it is still marked as a 'potential bug'. Does that mean there is need for more proof, or more people who experience this problem?

jbrazio, 2016-05-04:
It's also marked as inactive, we were waiting for you feedback.