MarlinFirmware/Marlin #3521: Z endstop not over bed at X0 Y0

Alex9779, 2016-04-16:
I have the Z endstop mounted on the print head but if the head is at the home position X0 Y0 the endstop cannot hit the bed.
Is there an option to use mesh bed leveling instead of auto bed leveling with the safe home option to home the head but for homing Z the head is move to a position where the endstop is over the bed?
For printing I could do a home of X and Y first then move the head and then home Z but mesh bed leveling is executing an auto home when starting and I do not find an option to move the head to a safe Z position.
Is there such an option or do I have to make a hack?

Alex9779, 2016-04-16:
Reviewing the code I realized that the G28 command procedure checks for Z_SAFE_HOMING with no other condition set.
That it is not set when using mesh bed leveling is that in the configuration Z_SAFE_HOMING is in the if block of auto bed leveling.
Does this have a special reason?
Do I have to expect any side issues if I just move Z_SAFE_HOMING out of the auto bed leveling so it is also respected when doing mesh bed leveling?

jbrazio, 2016-04-16:
I believe it should be the other way around, ABL is stable and MBL should follow and obey the same rules. There is a dedicated topic about this at MarlinDev.

Alex9779, 2016-04-16:
Ok @jbrazio I am irritated.
I do not find the top on MarlinDev, can you post a link?
And sorry if I bug but can you detail you answer a bit about why ABL is stable and which rules MBL should obey? Or maybe the topic on MarlinDev which I do not find covers my questions?

I did some experiments and have created a branch where I am able to compile using MESH_BED_LEVELING and Z_SAFE_HOMING here:
https://github.com/Alex9779/Marlin/commit/85f06b3aa6566385a6d8819eec900cf3b836a798

Moved the things for Z_SAFE_HOMING out of the auto level, had to comment a sanity check (didn't bother to change it) and have to use a different feedrate when doing safe homing because it was set to something that is obviously only set when using auto bed levelling. Just copied something from above...

I tried this on my machine and it seems to work. I can mesh level and the homing is safe. Didn't try to print yet...

jbrazio, 2016-04-16:
Never mind, mea culpa.
The topic is right here at the repo, sorry for the confusion. Have a look at #3488.

Alex9779, 2016-04-16:
Ahh ok found it, read it, got some points I think.
Now I understand your point of view.
Still I wanna use MBL because the experience other users made with the IR probe we have on the printer varies so ABL is not so reliable.

I am going to do some more testing with my hack and see if it works when printing...

jbrazio, 2016-04-16:
ABL is reliable in some situations, uneven bed is not one of them for sure.

What IR probe are you speaking of ?

Alex9779, 2016-04-16:
I was a backer of the E3D Kickstarter campaign for the BigBox.
It is this IR probe: https://miscsolutions.wordpress.com/mini-height-sensor-board/

Alex9779, 2016-04-16:
This whole issues came up for me with the new Titan extruder and the new hybrid (direct and bowden for dual printing) carriage E3D released last weekend...

thinkyhead, 2016-04-17:
We have been waiting to get ABL stabilized before making `Z_SAFE_HOMING` a standard feature, since it is currently, in part, integrated with ABL. And I wouldn't mind being able to home Z first, then do a raise, then home XY, but currently I can't do that. So there's much yet to do to make homing fully configurable for all setups.

Alex9779, 2016-04-17:
@thinkyhead catch me up on this, you make it a feature request but you want a fix for this for RC6?
For the moment the only "legal" option seems to be to use ABL.
I did not do a print yet with my hack, had to do some restructuring of my repos yesterday.
But homing and leveling worked pretty well.
So I see no reason why a print shouldn't do because what I did only affects homing...

thinkyhead, 2016-04-18:
I don't think it's wise to make `Z_SAFE_HOMING` a general feature for 1.1.0 right now. There will be too much testing involved, and we've only just stabilized the leveling and homing code without it. To put it another way, I personally don't want to open that can of worms right now. I think it will have to wait for 1.1.1.

thinkyhead, 2016-04-20:
We have a Feature Request topic for Z_SAFE_HOMING as a general feature, and I know we will do it very soon after 1.1.0 – so that it can be in 1.1.1. So… I will close this particular topic.

closed 2016-04-20 by thinkyhead