MarlinFirmware/Marlin #3523: M109 hanging when previously interrupted

JFD-Mtl, 2016-04-16:
Hi, I recently began updating the Marlin on my Printers (previous: 1.0.2, new: current Marlin, constantly updated by git pulls)

I have noticed that whenever, during the start of a print, I interrupt a M109 (set and wait for hotend temp) by using "stop print" on my controller, when I restart the print, the print never takes off and gets stuck at the M109. Then, If I do something to bring the hotend outside of the "close temperature range", let's say by manually extruding a lot of material, the M109 kick in and the print start.

so, to resume, I get this behavior when:
Start a print:
   - Set Temperatures (M190, M104)
   - Zero all axis
   - Wait for Hotent Temperature (M109)
Stop The print before the temperature has been reached (note: set temperatures stays set )
Restart the print
   - Set Temperatures (M190, M104)
   - Zero all axis
   - Wait for Hotent Temperature (M109)
   - It then waits for a very long time

I do not get this behavior when:
Manually set the temperatures
Wait until the temperatures are stable
Start a print:
   - Set Temperatures (M190, M104)
   - Zero all axis
   - Wait for Hotent Temperature (M109)

It also happened at least once when I stopped a print while it was printing it's first layer.

Thanks

jbrazio, 2016-04-16:
> Wait for Hotent Temperature (M109) Stop The print before the temperature has been reached (note: set temperatures stays set ) Restart the print

This only happens when you stop the print with temp very close to target temp or this happens when you stop the print at any given temperature ?

JFD-Mtl, 2016-04-16:
it doesn't seem to change much wether I interrupt the M109 when the temperature it's still climbing or during it's residency period.

JFD-Mtl, 2016-04-17:
It does show this behavior on the two printers I recently flashed with the current Marlin. But I must add that I can't reproduce this every single time. I guess there are some parameters influencing the bug that I still don't understand.

thinkyhead, 2016-04-17:
Sounds like we might need to reset the residency timer when someone cancels heating – e.g., by using Stop Print from the LCD.

thinkyhead, 2016-04-17:
@JFD-Mtl What values are you using for `TEMP_RESIDENCY_TIME` and `TEMP_HYSTERESIS`?

JFD-Mtl, 2016-04-17:
The same I have been using for over a year on all my 6 printers.

`#define TEMP_RESIDENCY_TIME 10  // (seconds)
#define TEMP_HYSTERESIS 3       // (degC) range of +/- temperatures considered "close" to the target one
#define TEMP_WINDOW     1       // (degC) Window around target to start the residency timer x degC early.`

thinkyhead, 2016-04-17:
@JFD-Mtl During heating there should be some temperatures logged to the serial console. If you happen to catch this in action, could you post the console output here for us to look at?

JFD-Mtl, 2016-04-17:
I usually run the printers in standalone. I will get this data Monday.
We have 6 printers. 4 of them running on Marlin. I only have problems on the two with the newer Marlin.

JFD-Mtl, 2016-04-17:
Can you be more specific on that part of the temperature log you need? I guess you need a sample of the console in debug mode while it hangs ?

thinkyhead, 2016-04-18:
You can sample temperatures with `M105` any time. But during heating with `M109`, without any debug mode required, temperatures will be printed to the console if one is attached. Just Connect from Pronterface before you start any standalone SD printing.

thinkyhead, 2016-04-20:
If this issue is fixed, I can close this topic, yes?

JFD-Mtl, 2016-04-20:
Ill reflash one of the printer with latest Marlin and keep you updated

thinkyhead, 2016-04-22:
Today's `RCBugFix` is better than yesterday's.

JFD-Mtl, 2016-04-23:
I tested it shortly with the current RCbugFix and it seems to be working properly.

Thanks

closed 2016-04-23 by JFD-Mtl