MarlinFirmware/Marlin #3531: Y-Axis running away during start of print

snowzach, 2016-04-17:
RCBugFix pull as of about 12:50AM EST 4/17 Cartesian, RAMPS, ABL and Safe Z Homing turned on... Everything pre-print seems okay (Homing, Bed Level, Display) but when I start a print the Y Axis takes off. It did start to print once but it was way over the side of the bed. It's almost like it thinks the middle of my bed (where I do safe Z-Homing) is 0,0 when it starts the print. Just speculating. Using GCODE that printed fine a couple days ago. For what it's worth it also hit my YMAX endstop and kept going. Anything I can do to help debug? I merged my config so trying to double check I didn't fat finger something.

jbrazio, 2016-04-17:
If you do the following what happens ?
```
G28
G29
G1 X0 Y0
```

snowzach, 2016-04-17:
That GCODE works as expected... This GCODE seems to cause the issue. I have tried a number of files and it happens intermittently but only when heated. I will debug more and report back. I just ran this code and it had the issue. Anything immediately off looking about my GCODE?
```
M107
M190 S105 ; set bed temperature
M104 S230 T0 ; set temperature
M104 S230 T1 ; set temperature
G28 ; home all axes
G1 Z5 F5000 ; lift nozzle

M109 S230 T0 ; wait for temperature to be reached
M109 S230 T1 ; wait for temperature to be reached
G21 ; set units to millimeters
G90 ; use absolute coordinates
M82 ; use absolute distances for extrusion
G92 E0
T0
G92 E0
G1 Z0.350 F7800.000
G1 E-2.00000 F2400.00000
G92 E0
G1 X114.905 Y106.102 F7800.000
G1 E2.00000 F2400.00000
G1 X115.314 Y105.693 E2.05255 F1800.000
```

jbrazio, 2016-04-17:
Usually this is the first thing at the start of the G-Code, before sending move instructions to the robot.
```
G21 ; set units to millimeters
G90 ; use absolute coordinates
M82 ; use absolute distances for extrusion
```

jbrazio, 2016-04-17:
You do have the origin correctly set on your slicer program right ?

thinkyhead, 2016-04-18:
We just patched a bug with `G92` today. Give the new `RCBugFix` code a test and see if it fixed the issue.

snowzach, 2016-04-18:
Okay, I was gonna say, just doing some debugging and requested a Z move and it flew off into the endstops in X...

snowzach, 2016-04-18:
Yah, so far so good.. I think that was the issue. Thanks! Closing.

closed 2016-04-18 by snowzach

thinkyhead, 2016-04-18:
With buffer overflows you never know what to expect, so I'm glad this is resolved!