MarlinFirmware/Marlin #3533: Man bed leveling is not working from LCD display

garaguido, 2016-04-17:
Hi , I used your latest Marlin RCbugfix ( Latest commit a545290 6 hours ago) and uploaded it with IDE 1.6.8 , here are the results :

    I can push the knob on "Prepare" and "Level bed" and then I turn it , notting happens , only the top row pixels from the text "Homing XYZ" is blinking (the text is not fully display missing top row pixels).
   The man bed level , still not working from my LCD , this was working when I uploaded with 1.0.6
    
    My LCD display is a RepRepDiscount Full Graphic Smart controller from Bigtree tech.com

thinkyhead, 2016-04-18:
I've been making changes to the menu manager to accommodate the kind of behavior MBL would like to have. It works fine on character-based controllers, but apparently there is some nuance I'm missing with graphical controllers. I just received a graphical controller, so I will do some testing with that over the next day or two and hopefully this will be worked out soon.

garaguido, 2016-04-18:
Thanks , I will look forward to it !!

thinkyhead, 2016-04-19:
@garaguido So far in testing with my shiny new graphical lcd controller I have not been able to reproduce the problem you describe. The knob seems  Everything seems to work fine for me. But perhaps there is some feature you have enabled which I don't that is creating this issue for you…? What does your `Configuration.h` look like?

Every day we fix a couple more bugs that can potentially cause issues like this. So be sure to give the latest `RCBugFix` a test each day to see if we accidentally fixed your issue.

Do you always see this glitch in Bed Leveling? It's possible it might fix itself after homing with `G28` or after doing a (small) test print. If the problem is always present (versus triggered by other things) then we might look for different culprits.

garaguido, 2016-04-19:
Thanks for the effort .

The **glitch **is not my mayor problem ,  the** MBL dos not work from my LCD** that is my problem.

The reason that I use MBL now , is that I have problems with the Auto BL .

ABL is always working fine by the first print , after when I start the printer the first time ( perhaps next day) the first Z layer is always perfect with Z-0.05mm. 

If I start a second or more prints , after a G29 (ABL) ,  then  there is a Z layer difference of 0.9 mm , then must I print ,for a good layer hight , with a Z -0.9mm difference .

With the `configuration.h` settings is notting wrong , I supose , all settings for my Prusa i3 is working fine with RC5.
My motherboard is a Xinray and compatible with Ramps 1.4

Here are my Configuration settings :

#include <U8glib.h>

/**
 * Configuration.h
 *
 * Basic settings such as:
 *
 * - Type of electronics
 * - Type of temperature sensor
 * - Printer geometry
 * - Endstop configuration
 * - LCD controller
 * - Extra features
 *
 * Advanced settings can be found in Configuration_adv.h
 *
 */
#ifndef CONFIGURATION_H
#define CONFIGURATION_H

#include "boards.h"
#include "macros.h"

//===========================================================================
//============================= Getting Started =============================
//===========================================================================

/**
 * Here are some standard links for getting your machine calibrated:
 *
 * http://reprap.org/wiki/Calibration
 * http://youtu.be/wAL9d7FgInk
 * http://calculator.josefprusa.cz
 * http://reprap.org/wiki/Triffid_Hunter%27s_Calibration_Guide
 * http://www.thingiverse.com/thing:5573
 * https://sites.google.com/site/repraplogphase/calibration-of-your-reprap
 * http://www.thingiverse.com/thing:298812
 */

//===========================================================================
//============================= DELTA Printer ===============================
//===========================================================================
// For a Delta printer replace the configuration files with the files in the
// example_configurations/delta directory.
//

//===========================================================================
//============================= SCARA Printer ===============================
//===========================================================================
// For a Scara printer replace the configuration files with the files in the
// example_configurations/SCARA directory.
//
//=============================================================================================
//====================== Prusa I3 printer met MKS board = Ramp1.4 =================
//=============================================================================================
[... 849 lines cut ...]
 **********************************************************************/
// Uncomment below to enable
//#define FILAMENT_WIDTH_SENSOR

#define DEFAULT_NOMINAL_FILAMENT_DIA 1.75  //Enter the diameter (in mm) of the filament generally used (3.0 mm or 1.75 mm) - this is then used in the slicer software.  Used for sensor reading validation

#if ENABLED(FILAMENT_WIDTH_SENSOR)
  #define FILAMENT_SENSOR_EXTRUDER_NUM 0   //The number of the extruder that has the filament sensor (0,1,2)
  #define MEASUREMENT_DELAY_CM        14   //measurement delay in cm.  This is the distance from filament sensor to middle of barrel

#define DEFAULT_NOMINAL_FILAMENT_DIA 1.75  //Enter the diameter (in mm) of the filament generally used (3.0 mm or 1.75 mm) - this is then used in the slicer software.  Used for sensor reading validation
#define MEASURED_UPPER_LIMIT         1.85  //upper limit factor used for sensor reading validation in mm
#define MEASURED_LOWER_LIMIT         1.65  //lower limit factor for sensor reading validation in mm
#define MAX_MEASUREMENT_DELAY       20     //delay buffer size in bytes (1 byte = 1cm)- limits maximum measurement delay allowable (must be larger than MEASUREMENT_DELAY_CM  and lower number saves RAM)

  #define DEFAULT_MEASURED_FILAMENT_DIA  DEFAULT_NOMINAL_FILAMENT_DIA  //set measured to nominal initially

  //When using an LCD, uncomment the line below to display the Filament sensor data on the last line instead of status.  Status will appear for 5 sec.
  //#define FILAMENT_LCD_DISPLAY
#endif

#include "Configuration_adv.h"
#include "thermistortables.h"

#endif //CONFIGURATION_H

thinkyhead, 2016-04-20:
I will continue to mull it over in my head and look at the code. If you are able to isolate a likely cause through testing and troubleshooting, or in any way figure out a way to make it behave better, please let us know.

garaguido, 2016-04-20:
Hi , Yes I will

garaguido, 2016-04-20:
Here , I will send you the two picture screens , from my LCD , about the issue I mensioned above.
The first screen is the standaard , the second screen is when I turn the knob , the top row pixels 
appears.
There is also a blinking pixel ( interval 1 sec ), in the right down corner. 
When I pushed the knob the pixel is gone and frozen .

[image]

Perhaps maybe this will help ?

PS: The new RCBugfix from 19-04-16 , I downloaded and installed ,was not yet the solution for me.

Blue-Marlin, 2016-04-20:
You might try
```@@ -943,14 +943,12 @@ void lcd_cooldown() {
       line_to_current(Z_AXIS);
       lcdDrawUpdate = LCDVIEW_CALL_NO_REDRAW;
     }
 
     // Update on first display, then only on updates to Z position
-    if (lcdDrawUpdate) {
-      float v = current_position[Z_AXIS] - MESH_HOME_SEARCH_Z;
-      lcd_implementation_drawedit(PSTR(MSG_MOVE_Z), ftostr43(v + (v < 0 ? -0.0001 : 0.0001), '+'));
-    }
+    float v = current_position[Z_AXIS] - MESH_HOME_SEARCH_Z;
+    lcd_implementation_drawedit(PSTR(MSG_MOVE_Z), ftostr43(v + (v < 0 ? -0.0001 : 0.0001), '+'));
 
     // We want subsequent calls, but don't force redraw
     // Set here so it can be overridden by lcd_return_to_status below
     lcdDrawUpdate = LCDVIEW_CALL_NO_REDRAW;
 
@@ -996,11 +994,11 @@ void lcd_cooldown() {
   /**
    * 4. MBL Display "Click to Begin", wait for click
    *        Move to the first probe position
    */
   static void _lcd_level_bed_homing_done() {
-    if (lcdDrawUpdate) lcd_implementation_drawedit(PSTR(MSG_LEVEL_BED_WAITING), NULL);
+    lcd_implementation_drawedit(PSTR(MSG_LEVEL_BED_WAITING), NULL);
     lcdDrawUpdate = LCDVIEW_CALL_NO_REDRAW;
     if (mbl_wait_for_move) return;
     if (LCD_CLICKED) {
       current_position[Z_AXIS] = MESH_HOME_SEARCH_Z;
       plan_set_position(current_position[X_AXIS], current_position[Y_AXIS], current_position[Z_AXIS], current_position[E_AXIS]);
@@ -1012,11 +1010,11 @@ void lcd_cooldown() {
 
   /**
    * 3. MBL Display "Hoing XYZ" - Wait for homing to finish
    */
   static void _lcd_level_bed_homing() {
-    if (lcdDrawUpdate) lcd_implementation_drawedit(PSTR(MSG_LEVEL_BED_HOMING), NULL);
+    lcd_implementation_drawedit(PSTR(MSG_LEVEL_BED_HOMING), NULL);
     lcdDrawUpdate = LCDVIEW_CALL_NO_REDRAW;
     if (axis_known_position[X_AXIS] && axis_known_position[Y_AXIS] && axis_known_position[Z_AXIS])
       lcd_goto_menu(_lcd_level_bed_homing_done);
   }
 
```
Hard to say how many dislpy updates we have then - since @thinkyhead  messed up the blink from per display update to per second.

garaguido, 2016-04-20:
The configuration file I showing was from earlear bugfixes a few days ago.

Blue-Marlin, 2016-04-20:
Update rate is a bit high with the above patch. However, we have no better things to do.

garaguido, 2016-04-20:
When the text on the display  "Homing XYZ" appears  after "Bed level", there is no action possible , also when I turning the knob .
The only thing I can do , is  resetting printer with -> power off.

Blue-Marlin, 2016-04-20:
There is no action required. The machine homes and when ready continues with "Click to Begin".

garaguido, 2016-04-20:
It stopped after "Homing XYZ" and there was no text "Click to Begin" . 

I now this and use MBL before ,but when I compiled and uploaded the version RC5 with Arduino IDE 1.0.6 the LCD with  MBL works then , but I can not print from the SD card LCD.

Perhaps this help you further ?

Blue-Marlin, 2016-04-20:
Did the machine home? Can you home it with a normal G28?
Sorry i'm just interested in the display problem.

garaguido, 2016-04-20:
Yes , I can homing with G28 or the command "Homing " on the display  . 

I can also using MBL without the LCD , direct from the Repetier GUI

Blue-Marlin, 2016-04-20:
Replacing all `axis_known_position` with `axis_homed` in ultralcd.cpp may also help (wrong criteria for homed).

garaguido, 2016-04-20:
Thanks , I will try it out

garaguido, 2016-04-20:
Thanks , great help !!! 

I have making the mods , compiled and upload it with the newest Arduino IDE  1.6.9_nightly and  this is working very well . 
There is still the one minor issue , the top row pixels of all the text lines and digits , after the command "Bed 
Level" is missing ( see the screen views above ).

Is there perhaps also a  solution in the future ?

Blue-Marlin, 2016-04-20:
The first patch was against the missing pixel lines. Have an other look, ether on the screen, or on the patch.

Getting stuck while waiting for homing was brought to light by your (rare) `#define DISABLE_Z true` setting. This is rarely tested.

garaguido, 2016-04-20:
I use " #define DISABLE_Z true" for the configuration , when  I use my Auto Bed leveling ( if its working properly see my explaining above )  , my contact from my cap.sensor is NO , and  therefore " true "

I use the newest RCbugfix from 19-04-16, and there are in this version no more
"@ -1012,11 +1010,11 @@ void lcd_cooldown() {  "    lines any more .

They all are already been deleted I supose ?

thinkyhead, 2016-04-21:
I have a pretty comprehensive patch now with #3579, which I will merge shortly. In addition to fixing graphical display issues and the `axis_homed` stuff as illustrated in AnHardt/Marlin#30, we now (finally) have a friendly display before each probe point showing your progress (`Next Point: 1 / 9`).

After I made these changes for graphical display, I didn't go back and check to make sure that character displays are unaffected. But they _should_ be! I will test with mine later.

Ultimately I want to make sure there are a minimal number of display updates (even though with this procedure it's not too important). Ideally I want the "new menu code" should do the same number of calls to draw as the "old menu code", but I'm only just untangling the way it was meant to work. Another comb-through of the code might yield another tweak or two.

thinkyhead, 2016-04-21:
Ok, I think I got it after myriad tweaks.

- Don't switch menu pages in the middle of a display update
- Don't change `lcdDrawUpdate` in a menu handler (unless a value changes without an encoder event and you want to update it now and for the rest of the draw loop)

To ensure the first, switching menu pages is now deferred until after the draw loop.

To ensure the second, I removed almost all changes to `lcdDrawUpdate` except to get the forced callback in `_lcd_bed_level_homing`.

I haven't been able to test these latest updates with my LCD but will do so as soon as I get home!

thinkyhead, 2016-04-21:
I was finally able to reproduce bad behavior, so I got into the code and patched up all the display issues I could locate. MBL should be more enhanced now also. I just submitted a patch, so please give the code a try as soon as you have a chance.

garaguido, 2016-04-21:
Hi , I used the latest " RCBugfix 6fac4d9 "from 21-04-16 en uploaded it , the LCD behave weird also there where no digits on the display for adjust Z hight , also de Y steppermotor has a verry slow behavier after MBL and Homing X Y Z .
I have used again the older version from 20-04-16 with the mods from Bleu Marlin its working fine by me

thinkyhead, 2016-04-21:
As it turns out #3579 didn't test too well for me, and all my attempts to work around the quirks of the menu handler kept hitting dead-ends. So I went back by one commit (only the last commit was untested) to the point where I had gotten it working as well as I could — without optimizing the redraws.

So the current code should work better.

thinkyhead, 2016-04-22:
@garaguido Please check the current code to make sure nothing was missed in the latest patch-up. If MBL is behaving better (with a couple of additional patches) then we can finally call this closed. (Hurray!)

garaguido, 2016-04-22:
thinkyhead ,  Hurray ....and thanks for your help !!

It is working verry well with the recent RCbugfix 21-04-16 and I find this also a verry good improvement for the position indicatior ....1/9 , 2/9 , etc... and indication "Bed level done" .

 I uploaded it with the Arduino IDE 1.6.9_nightly , without any problem was perfect from first time!!!

But , I found another minor bug , using the LCD display and print from SD card  , must I make a new issue for it ?

closed 2016-04-23 by thinkyhead