MarlinFirmware/Marlin #3534: question about g28 after g29

jfpion, 2016-04-17:
It look like a g28 x y cancel the autoleveling by g29
normal ?

is there a way to avoid this canceling (keep this behaviour only with g28 alone or g28 z )

whan you stop to change the filament (colors) it is difficult not to move the x carrige and lose step so you need to do a g28 x before restarting the print

thank you

jbrazio, 2016-04-17:
This behavior is by design.
Actually do to the error margin each physical endstop inherently has doing a `G28` in mid print seems not to be advisable. When changing filament if the steppers are powered then it's kinda difficult to "loose steps" or move them around, unless you're really trying to do it.

Roxy-3DPrintBoard, 2016-04-17:
> whan you stop to change the filament (colors) it is difficult not to move the x carriage and lose step so you need to do a g28 x before restarting the print

You should be using something like:   M600 Z50 to change filament.   It will lock the X & Y motors but unlock the filament extruder motor.   You won't accidentally cause the nozzle to move with a missed step.

jfpion, 2016-04-17:
@jbrazio i did a lot mixed color print before using the autolevel with an g28 x at each change. never got a pb nor a visible mis alignment in x. and yes it is not really easy to lose step but it could happen and if you can't make a g28 x the part is definitly lost. so why this design ? i can understand for the g28 z limitation but in y and z there no real reason, it's better not to have to use it but it is not going to mess with the autolevel.

@Roxy-3DPrintBoard the pb is not the extruder motor as I use direct extruder, the pb is the presser lever which could be hard to move and make the x carriage to move when open or close it

thank you two

jbrazio, 2016-04-17:
> so why this design

There is a probability to move things around indeed.. but when motors are powered on (`M600`) it's odds are very low when compared with the probability of having additional error due to repeatability issues of the endstops when homing mid print (and I will not even speak of the possible physical constrains of trying to home with plastic on the table). Try for yourself, play a bit with the `Z_MIN_PROBE_REPEATABILITY_TEST` configuration flag and `M48`.

jfpion, 2016-04-17:
M48 Z-Probe Repeatability test
Mean: 1.479118
Standard Deviation: 0.013960
not too bad for a mechanical switch ...

is there the same for the x and y switch ?

thank you

jbrazio, 2016-04-17:
Unfortunately this test only applies to Z endstop, but this is a good remark, we should have the same test for all the endstops.

For comparison, this is the output of an induction sensor:
```
Mean: 0.883275
Standard Deviation: 0.004564
```

jfpion, 2016-04-17:
not the same league than my old switch. anyway i've got a glass bed so i need to go for a capcitive sensor, is it the same accuracy ?

and i think using the same logic than the z sensor could be quite easy juste need to add the letters for the endstop to be tested (why not a test for the 3 endstops ?)

thank you

closed 2016-05-06 by jbrazio