MarlinFirmware/Marlin #3535: Request - automatically calculate Z_PROBE_OFFSET_FROM_EXTRUDER on delta

zylantha, 2016-04-17:
Currently Z_PROBE_OFFSET_FROM_EXTRUDER must be manually entered into Configuration.h to set the vertical height at which the probe will be triggered, which is then later used to calculate actual zero points after autolevel, however there is little documentation on how to derive this value manually.

On a delta printer, this value could actually be calculated very simply as follows:

1. User sets MANUAL_Z_HOME_POS at centre of print bed as per standard recommendation (office paper test)

2. At commencement of autolevel G29, the first probe should be at centre of the print bed (where MANUAL_Z_HOME_POS was set).

3. The difference between that first probe position and MANUAL_Z_HOME_POS will be the Z_PROBE_OFFSET_FROM_EXTRUDER - and can be used on all other probe points.

This method should work for MECHANICAL_PROBE (+ve value) - and would probably work for other types of probes as well (-ve value).

Roxy-3DPrintBoard, 2016-04-17:
You are making an assumption that the position of the nozzle is accurately known once the Delta is homed.  Many people use several different pieces of glass for their prints and they vary in thickness.   So in that case it is still necessary to know the distance from the probe firing to the nozzle height.

zylantha, 2016-04-18:
That is a fair point, and one that I understand Smoothieware doesn't provide for.

However making this an optional 'automatic' measurement would make it easier for those setting up new printers that did not have multiple print beds, without affecting those that want to set it manually.

Alternatively, providing a command to measure this on request, and store it  in EEPROM would also be neat and cater for multiple use cases.

thinkyhead, 2016-04-19:
@zylantha I still don't get your logic here. You wrote:
> 2. At commencement of autolevel G29, the first probe should be at centre of the print bed (where MANUAL_Z_HOME_POS was set).

No. The first probe won't be in the center because that's where the nozzle is after homing. The printer must move the probe over to the center if it wants to probe the center. How does the printer know how far to move the probe so that it is in the center? By using the probe offsets that were given in the configuration.

> 3. The difference between that first probe position and MANUAL_Z_HOME_POS will be the Z_PROBE_OFFSET_FROM_EXTRUDER - and can be used on all other probe points.

It sounds like what you want is a procedure like this:
- User chooses "Measure Probe Offset" from the menu.
- The printer homes to put the nozzle in the center.
- The printer probes straight down to the bed, nozzle still in the center.
- The printer moves the probe to "the center" (using the existing probe offset).
- User is asked to move the probe with the LCD until it is at the exact center.
- When done the user clicks the controller, and the offset is saved to EEPROM.

closed 2016-05-06 by jbrazio