MarlinFirmware/Marlin #3536: Time the given temperature must be held until print starts

yellobello, 2016-04-17:
Hi!

Apparently, when printing from SD (maybe when printing off serial as well) the set temperature must be held for a certain time until the print starts. Since my temps are some time off for 1°C, the print at the moment never starts. (when the bed is ok, the extruder is not and vice versa.
I had a previous marlin dev version where this problem did not exist.

Where in the code is the value how long to wait?

I coud not find it.

p.s. yes, i did PID tune

jbrazio, 2016-04-17:
Could you test the latest **RCBugfix**, there you have some options which allows you to fine tune your nozzle and bed temperature management.

```
// Extruder temperature must be close to target for this long before M109 returns success
#define TEMP_RESIDENCY_TIME 10  // (seconds)
#define TEMP_HYSTERESIS 3       // (degC) range of +/- temperatures considered "close" to the target one
#define TEMP_WINDOW     1       // (degC) Window around target to start the residency timer x degC early.

// Bed temperature must be close to target for this long before M190 returns success
#define TEMP_BED_RESIDENCY_TIME 10  // (seconds)
#define TEMP_BED_HYSTERESIS 3       // (degC) range of +/- temperatures considered "close" to the target one
#define TEMP_BED_WINDOW     1       // (degC) Window around target to start the residency timer x degC early.
```

If this does not solve your issue then we may need to dig deeper.

yellobello, 2016-04-17:
It does the Job, thank you!  

I lowered it to 5 sec and its fine now, maybe a default of 5 would be better?
Since when is this in the code? is it fairly new?

jbrazio, 2016-04-17:
The options controlling the bed are really recent but the nozzle ones are there for a while now.

thinkyhead, 2016-04-18:
I know 10 seconds isn't all that long, but I think 5 seconds would be better – and pretty reasonable, assuming a narrow enough window and hysteresis.

closed 2016-05-04 by jbrazio