MarlinFirmware/Marlin #3549: Z axis moves when using "G92 E0"

Alex9779, 2016-04-18:
I was running a script to calibrate my extruder E steps with RC3.
Did not try it with RC4 or RC5 though now with the RCBugFix of two days ago I have an issue.
Here is the script:
`
;config
G21 ;metric
G90 ;absolute positioning
M83 ;relative extrusion

;home
M117 Homing Axis.
G28

;raise z
M117 Positioning Bed.
G1 Z4 F200

;dock
M117 Docking Printhead.
T0
G1 X100 Y200 F3000
G1 X100 Y235 F3000

;extrude 10mm
T0
G92 E0
G1 E10 F200
G92 E0

M117 10mm Extruded.
;BigBox Calibration Code End
`
When I print this from SD, did not try from a host, the bed begins to move when the "G92 E0" is executed and moves and moves. Did not let it move to the bottom so I don't know if it will respect the limit or not.

Maybe this has something to do with the "M83" I use

fbarcenas, 2016-04-18:
i am also have this problem, after extruder the first line do G92E0 and the Z go up and broken extruders...

snowzach, 2016-04-18:
Pull the latest, this was just fixed last night I believe. I had the same issue except mine was X-Axis

fbarcenas, 2016-04-18:
Yes i have this

Alex9779, 2016-04-18:
Going to test RCBugFix of last night later...

Alex9779, 2016-04-18:
Tested with latest RCBugFix, seems to be fixed now...

closed 2016-04-18 by Alex9779

thinkyhead, 2016-04-19:
Others can confirm this issue is solved?