MarlinFirmware/Marlin #3551: Babystepping values

sarf2k4, 2016-04-18:
I would like to have clarification on how babystepping works especially when the knob turned, does the adjustment based on each step per knob turned or based on microns?

As in steps per knob turned
```
each knob click = 4 values = 4 steps
```

If so, can I request to display values when babystepping adjustment are made? Just use a relative adjustment would do and every time user exits the babystepping, it will reset from 0 again

thinkyhead, 2016-04-19:
> adjustment based on each step per knob turned

```cpp
encoderPosition += (encoderDiff * encoderMultiplier) / ENCODER_PULSES_PER_STEP;
. . .
int distance = (int32_t)encoderPosition * BABYSTEP_MULTIPLICATOR;
. . .
babystepsTodo[axis] += distance;
```

> can I request to display values when babystepping adjustment are made?  

Something like this? #3566

sarf2k4, 2016-04-19:
@thinkyhead , yes, like #3566 

A little bit of off topic, how can I change the knob resolution to 1 turn = 1 value? Currently I'm using RRD 2004 and each turn would have 4 values at once.

Blue-Marlin, 2016-04-19:
```
//
// ENCODER SETTINGS
//
// This option overrides the default number of encoder pulses needed to
// produce one step. Should be increased for high-resolution encoders.
//
//#define ENCODER_PULSES_PER_STEP 1
```

thinkyhead, 2016-04-20:
As blue-marlin points out, yes, adjust the relevant settings in the LCD section of `Configuration.h`. Cheers!

closed 2016-04-20 by thinkyhead